OctoPrint/OctoPrint #3693: [Request] Report g-code to fast for printer to user

hazelybell, 2020-08-18:
I'm having a lot of issues with pauses due to the buffer draining and the printer pausing, extruding a puddle, and then resuming without nozzle pressure. I have concluded, in agreement with the discussion in #3209, that the ADVANCED_OK P-values are fairly useless to Octoprint for asynchronous command buffering, since they are already made asynchronous (Marlin reports ok as soon as the g-code is in the move block buffer, regardless of whether the moves is completed.) 

Theoretically, Octoprint could use the ADVANCED_OK P-value to automatically adjust Marlin's M205 B-setting to increase slowdown, but the slowdown implementation in Marlin is broken for several reasons.

However, it may be possible for Octoprint to detect P-values bouncing all over the place and report that to the user, explaining to them that they should decrease their print feedrate, or their feedrate %, or their accelation settings, or their jerk settings, or their slicer settings to produce longer moves.

I believe an informative message such as this would be very helpful to many users with pause-blobbing problems caused by planner buffer underruns.

I don't think asynchronous command sending could help that much. It might help a little. But fundamentally the problem is that the CPU just can't plan fast enough for the feedrates set in the file. 

Marlin's slowdown functionality should automatically fix this. However, 

- It's not always enabled and there's no way to tell if it's enabled.
- M205 B is allowed (and stored in the EEPROM) even if slowdown is disabled, but it has no effect.
- The slowdown code only runs on move insertion, and if they buffer is full, does nothing. This means that it can insert a lot of fast that should be slowed down as long as they're in line behind slower moves. Then, once the slower moves are finished the buffer empties before a single new slow move can be inserted. If the buffer empties completely, Marlin intentionally stalls until the buffer's full again, making the stall even longer!
- It may be better to do any automated adjustments to settings like M205 B in the firmware, so they can fix the same issue if it occurs on an SD-card print.
- The best solution is to just not use slicer settings that my 16Mhz controller can't handle in the first place.

HELEMAALNIKS, 2020-08-20:
I don't really know how g-code works, but I'm experiencing the same issues that my printer will sometimes just pause randomly...

chendo, 2020-10-09:
I'm having a crack at implementing gcode in Marlin that enables interval reporting of buffer stats (tentatively `M576`), which will expose the planner and command buffer stats as well ashow times the command buffer underrun since the last report, which combined with an Octoprint plugin, should be able to surface the amount of times the command buffer ran out to the user.

Reporting on an interval should decrease the amount of serial comm load as `ADVANCED_OK` recommends increasing `TX_BUFFER_SIZE` to fit, too.