OctoPrint/OctoPrint #3694: Reverse proxy is not working properly.

aa15032261, 2020-08-20:
#### What were you doing?

I'm trying to setup a reverse proxy in nodejs using node-http-proxy, however OctoPrint's reverse proxy isn't fully working. I noticed that the "X-Script-Name" header is able to change the base path of the website, however, everything start with "/BASE_PATH/static/webassets/packed_" and "/BASE_PATH/sockjs/" will throw error 404. I have to remove the base path for these two types of request to make the reverse proxy working.

```
let proxyPath = req.url;
if (proxyPath.startsWith('/BASE_PATH')) {
    proxyPath = proxyPath.replace('/BASE_PATH', '');
}

proxy.on('proxyReq', function (proxyReq, req, res, options) {
    proxyReq.setHeader('X-Script-Name', '/BASE_PATH');
    
    if (
        proxyPath.startsWith('/static/webassets/packed_') ||
        proxyPath.startsWith('/sockjs/') 
    ) {
        proxyReq.path = proxyPath;
    }
});
```

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint
1.4.2

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

aa15032261, 2020-08-20:
The base path is not reflected on webcam stream as well, it will return 404.

fieldOfView, 2020-08-20:
The webcam stream is not part of OctoPrint. mjpegstreamer is just a bundled component of Octo*Pi* (not OctoPrint).

foosel, 2020-09-03:
I need way more information on what you are doing here and how you are trying to access which endpoints. Are you going through haproxy or against OctoPrint directly? What's the exact requests going out, what's the responses you are seeing. Also I need at the very least `octoprint.log`, your JS error console in the browser and ideally an example client code with which to reproduce.

aa15032261, 2020-09-05:
For example, the login page has 14 requests:
[image]

```
/
/static/css/bootstrap.min.css
/static/css/bootstrap-responsive.min.css
/static/vendor/font-awesome-4.7.0/css/font-awesome.min.css
/plugin/loginui/static/css/loginui.css
/static/js/lib/jquery/jquery.min.js
/static/js/lib/sockjs.min.js
/static/js/lib/bootstrap/bootstrap.js
/static/js/lib/lodash.min.js
/static/webassets/packed_client.js?c542e09b
/plugin/loginui/static/js/main.js
/static/img/tentacle-32x32.png
/sockjs/info?t=1599336809028
/sockjs/285/v1aphxsd/websocket
```

From my understanding, 'X-Script-Name' header is supposed to change the absolute path of the octoprint website, however, some pages will not load with custom absolute path.

For example, /sockjs/info works fine without custom base path:
[image]

But if I use a custom base path with 'X-Script-Name' header, the page will throw 404
[image]

These urls also disrespect the X-Script-Name header:
/api/files/local
Everything under "/static/webassets/" and "/sockjs/"

foosel, 2020-09-07:
> From my understanding, 'X-Script-Name' header is supposed to change the absolute path of the octoprint website, however, some pages will not load with custom absolute path.

You are making a wrong assumption about what a reverse proxy actually does, what its role is in the whole communication and what the proxied application does in response.

All this `X-Script-Name` header will do is make sure that any *URLs generated by OctoPrint* into HTML code and/or API endpoint responses will contain the provided base path. It will *not* make the whole OctoPrint server be magically re-located to a different base path. It's the task of a [properly configured reverse proxy](https://community.octoprint.org/t/reverse-proxy-configuration-examples/1107) to rewrite incoming requests to `<reverse proxy host>/toms-ender-3-pro/sockjs/info` into `<octoprint host>/sockjs/info`. For OctoPrint, the requests will still target `/sockjs/info`, it's the job of the reverse proxy to do the address translation and forward the translated request to OctoPrint, which will then return sensible responses which the proxy then forwards back to the client. At no point will OctoPrint itself actually respond to the base path of `/toms-ender-3-pro`.

So - based on the information you've provided here, everything is working as expected and it's your reverse proxy that needs some fixing. I'm thus closing this.

closed 2020-09-07 by foosel