OctoPrint/OctoPrint #3695: Octoprint paused randomly

luco85, 2020-08-20:
Hello to everyone
After update octoprint to lastest release i have some issue. 
It pause randomly. no error in terminal. I not understand why 
I attach my log of last print
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5103287/octoprint.log)

cp2004, 2020-08-20:
Hey @luco85 
Seems like your issue is with Klipper firmware, as that is where the errors in the logs are coming from.
This issue tracker is for bugs & requests for OctoPrint, and OctoPrint only, if you have an issue with your setup, please try and get help in the [community forum](https://community.octoprint.org).

luco85, 2020-08-21:
Ok thanks! So I check klipper’s log
In octoprint log there is no error about my issue right?