OctoPrint/OctoPrint #3704: JS-Exception handling not working with disabled ErrorTracking Plugin

OllisGit, 2020-08-25:
#### What were you doing?

During development of a new Plugin sometimes I use wrong KnockJS Bindings.
E.g. 
```
viewModel.myProperty = ko.observable();
...
<span data-bind:"text: viewModel.mmmProperty">
```
Expected behaviour in the development console of the browser:
```
...ReferenceError: Unable to process binding "text: function(){return viewModel.mmProperty }"...
```
I disabled the ErroTracking-Plugin. Now I always receive a not helpful error message:
```
...ReferenceError: Sentry is not defined...
```

**Reason:** 
https://github.com/OctoPrint/OctoPrint/blob/devel/src/octoprint/static/js/app/main.js#L681
```
} catch (exc) {
	if (Sentry) {
```
The variable Sentry doesn't exists.

**Solution:**
Replace ```if (Sentry){``` with ```if (typeof Sentry !=='undefined') {``` 

see previous description

#### What did you expect to happen?

"Real" Error description should appear in the Browser console

#### What happened instead?

An other exception occurs, because the error-handling had an error ;-)

#### Did the same happen when running OctoPrint in safe mode?

No, because my custom plugins were disabled.

#### Version of OctoPrint

Latest, 1.4.2. in OP 1.3.10 it is working

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

ANET E10

#### Browser and version of browser, operating system running browser

Chrome V84

#### Link to octoprint.log

Not needed, because it is a client side issue.

#### Link to contents of terminal tab or serial.log

Not needed, because it is a client side issue.

#### Link to contents of Javascript console in the browser

See above

#### Screenshot(s)/video(s) showing the problem:

See above

I have read the FAQ.
Yes

cp2004, 2020-08-26:
@OllisGit Tried this and it does block the whole UI from running. Haven't had a chance to test out your fix, but would you like to make a PR, seeing as you've done the hard work finding it :) ? If not, and I can try your fix and add it to the PR stack for when Gina comes back next week.

OllisGit, 2020-08-27:
@cp2004 fyi: PR is created: https://github.com/OctoPrint/OctoPrint/pull/3706

Thx, for looking into
Olli

foosel, 2020-09-03:
And merged, thanks! Fix be released with 1.5.x.

foosel, 2020-11-30:
1.5.0 has just been released.

closed 2020-11-30 by foosel