OctoPrint/OctoPrint #3705: Watchdir Does Not Recognize Files [Renamed?]

ciordia9, 2020-08-25:
At first I thought the watchdir was not recognizing large exports as it will catch small ones. Then I noticed there was a tmp file being created in the watchdir and being renamed at the end of the process which makes me believe: 

Watchdir does not recognize file names renamed within rapid succession of export. 

**Issue explanation:** This behavior has been seen with [Ideamaker](https://www.raise3d.com/ideamaker/) and might exist with other slicers that first export their files to a temporary filename within the path of the watchdir and then on completion rename it to the destination file name. This confuses the watchdir watchdog and will never get picked up by Octoprint. 

- Polling directories does also not seem to fix this. 
- Upon rebooting of the device/Pi the file is picked up at that point.

I've attempted to watch the logs (debug priority) but I don't see anything relevant.

Small files <200kb seem to not get hit by anything > seems to get caught.

cp2004, 2020-08-25:
Hey @ciordia9 - Please provide the `octoprint.log` here, it may not *look* like there is something in there that helps, but it contains important information about the underlying environment to help reproduce the issue. Thanks.

ciordia9, 2020-08-25:
> Hey @ciordia9 - Please provide the `octoprint.log` here, it may not _look_ like there is something in there that helps, but it contains important information about the underlying environment to help reproduce the issue. Thanks.

How much of it do you want, clear, fresh boot, test load? There are literally no new-lines formed when this occurred but I get it when you speak of the rest of the architecture.

I have at least one other that has reported this in our 3dp discord. I'll see if I can get them to post up as well.

cp2004, 2020-08-25:
If you could test load the file into the watchdir, then just download & drop here the octoprint log, so that anyone who goes to reproduce it can compare details etc. Don't worry if it's a bit long with previous events - I have a good scroll wheel 🙂

ciordia9, 2020-08-25:
[octoprint.report.log](https://github.com/OctoPrint/OctoPrint/files/5125803/octoprint.report.log)

I've edited the log to show where the file was uploaded..(CIORDIA DEBUG)  You'll see nothing.
Just to showcase that it does indeed work if you copy the file over after the fact annotated right behind it is an actual watchdir firing off.

cp2004, 2020-08-25:
Thanks for that, @ciordia9, I can take a look but no guarantees that I'll find anything, since I'm only triaging some tickets while Gina (foosel) is on a break. Just wanted to make all info is there.

ciordia9, 2020-08-25:
All good. I've been working around it for a good while. Just ran into others having the same problem and explained what I had found and thought I should probably stop just Discord chatting and actually write it up. :)

foosel, 2020-09-03:
What's the name of the temporary file that gets exported first? Does it work if you enable active polling under Settings > Folders?

ciordia9, 2020-09-05:
> What's the name of the temporary file that gets exported first? Does it work if you enable active polling under Settings > Folders?

Hi Foosel, no, I keep active polling on and it never picks it up (until reset.) 

I attempted it again and had to screenshot the passing file as you don't see it if you blink--

[image]

qt_temp_JL2099

That gets renamed into a proper filename whatever.printme.gcode

I cleared the program and loaded a few models and exported and caught a variation

qt_temp.JL3449

foosel, 2020-09-07:
Turns out the watchdog has separate events for file added and file moved/renamed. So far it only triggered on creation. When a temporary file is first created in the folder by your slicer it won't match the pattern matcher, and the subsequent name change will only trigger a move event and not be detected, thus the file never gets added. 

Easy fix, just needed to make the watchdog aware of file renames as well. Pushed to `maintenance`, to be released with 1.5.0.

commit 497cf925 (2020-09-07): 🐛 Fix renames not triggering watched folder

Closes #3705
closed 2020-11-30 by foosel