OctoPrint/OctoPrint #3708: gcode editor not updating in virtual environment

xeddog, 2020-08-28:
#### What were you doing?
I have 3 RPi3B+, each dedicated to a separate printer.  Two of the three rarely ever get shutdown or rebooted, while the third one gets powered down whenever the printer is shutdown (usually nightly).  One morning when I turned that printer on, the Pi obviously did too, and when it was ready there was an update for the gcode editor plugin to 0.2.9 available.  I installed it and everything seems to be fine.   This Pi is running Octopi 0.17.0, Octoprint 1.4.2.
For the other two Pi's, I used the software update function to get the gcode editor update for those two as well.  But there was no notification that an update was available, and the gcode editor plugin (v 0.2.8) was still blacklisted on both of the Pi's.  I tried using the FORCE check but that had no effect.  It has been several days and still no update on those two pi's.   These are running Raspian GNU/Linux 10 (buster), and Octoprint 1.4.2.
The only other difference is that the two that did not get the update for gcode editor are running Octoprint in a Virtual environment.  

Also maybe worthy of note is that maintenance on all three instances of the OS are up-to-date.

1. Boot Pi
2. Click on the settings wrench icon
3. Scroll down to Software Update
4. Scroll all the way to the bottom of the list
5.  Click "force check for update"
6.  Everything is up-to-date panel is displayed
7.  Plugin Manager still shows
[image]

#### What did you expect to happen?
I expected the gcode editor update to be available to all three instances, and installable.
#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

Safe Mode can't be used because all plugins are disabled.

#### Version of OctoPrint
1.4.2

#### Operating System running OctoPrint
1.  OctoPi 0.17.0  (This one works)
2 and 3 Raspian Buster with current maintenance  (These two do not)

#### Printer model & used firmware incl. version
Printers are:
1 - CR-10S running Marlin bugfix-2.0.x on a MKS SGEN L controller.  Marlin is current as of approximately 8/20 give or take
2 - Ender 5 Plus running same marlin on a SKR Pro V1.1 controller.
3 - Monoprice Mini Select running same Marlin on a SKR V1.3 controller

Browser is Firefox V80.  Host machines are 1 Windows 10nPro laptop, and two Ubuntu 20.04 Linux machines.
#### Browser and version of browser, operating system running browser

#### Link to octoprint.log
Octoprint log isn't available but I will attempt to get one if needed.

jneilliii, 2020-08-28:
I think this is because the GCode Editor plugin is on the blacklist possibly for the latest version of OctoPrint.

xeddog, 2020-08-29:
> 
> 
> I think this is because the GCode Editor plugin is on the blacklist possibly for the latest version of OctoPrint.

So why did the one instance of Octoprint that is not in a virtual environment UN-blacklist the gcode editor and do the update, and the two that are in a virtual environment did not?  The two remaining instances of Octoprint **should** update.  I also asked this question on one of the Facebook groups, and it sounds like one or two people there were able to update, and they did not have the virtual environment.

jneilliii, 2020-08-29:
Maybe the blacklist is disabled on the ones that worked? Looking at the releases it seems that the latest 0.2.9 version shouldn't be blacklisted. You could manually install copying/pasting the URL below in plugin manager > Get More > ...from URL.

```
https://github.com/ieatacid/OctoPrint-GcodeEditor/archive/0.2.9.zip
```

cp2004, 2020-08-29:
Try manually installing the plugin, as above. All of your instances above will be virtual environments as that's how OctoPi comes installed, so that's unlikely the issue.
Blacklist processing happens on server startup, so maybe it does not recognise the new version until you have restarted? Just a thought, I don't have the plugin installed so don't know exactly how that's going.

xeddog, 2020-08-29:
> All of your instances above will be virtual environments as that's how OctoPi comes installed, so that's unlikely the issue.
> Blacklist processing happens on server startup, so maybe it does not [...]

The instance that DID install is an OctoPi that is not in a virtual environment.  This is also the oldest of the instances and was originally installed before venv came along, and has been upgraded along the way.  I have been thinking about re-imaging all of them just to have commonality among all of them, but I don't want to mess with this one since it works.  

In trying to get all of this worked out, I did reboot both of the ones that do not update, and when that did nothing, I turned them off completely (removed the power), let them sit for a few minutes, and then brought them back up.  Still nothing.

Regarding a manual install, I have thought about doing that too, but would that bypass any blacklist processing that might be needed later on?

Wayne

jneilliii, 2020-08-29:
Manually installing won't effect the blacklist processing. The only thing you risk is that the plugin you manually install causes OctoPrint to not load properly.

cp2004, 2020-08-29:
>The instance that DID install is an OctoPi that is not in a virtual environment. This is also the oldest of the instances and was originally installed before venv came along

Ok that's quite old then? I saw this bit in your initial post, maybe you mean a different version.
>Octopi 0.17.0

Plugins installed manually still get processed by the blacklist, on restart. You can [see the list here if you want](https://plugins.octoprint.org/blacklist), just out of curiosity.

Just tried this myself, and I can't get it to update either. Installed, then it is blacklisted, so doesn't load (I'm running the branch maintenance, so `1.5.0.dev127+gbd7ed94b4`). I think this means that the update information doesn't get loaded either - as that requires calling `get_update_information()` on the plugin, but it is not loaded so hence this is not possible. Limitation of the blacklist/OctoPrint's plugin system in this case then AFAIK.

The instance that did update, by any chance, was running OctoPrint 1.4.0 or earlier? In that case it would not be blacklisted, and then update successfully. Only way around this is to manually install new version.

In some ways, I suppose this is a good idea, since then it limits the rollout - what if the plugin author released a new version, but it still broke OctoPrint? Users would come here and tell *us* that OctoPrint broke. Then we can blacklist the new version, and tell the author what's up. It is possible to add plugin notices in the repository, so maybe that is a good course of action if it's verified the new version works, to add a notice to say 'hey, new version fixed, update manually'.

jneilliii, 2020-08-29:
This shouldn't be blacklisted with the latest version, because 0.2.9 is not flagged in the blacklist.json. I don't want to mess with the announcements/blacklist stuff until Gina gets back.

cp2004, 2020-08-29:
Not a problem, it can wait I guess. The issue here is that the update information isn't read from 0.2.8 *because* it is blacklisted so not run. Requires a manual reinstall of 0.2.9.

xeddog, 2020-08-30:
> 
> 
> > The instance that DID install is an OctoPi that is not in a virtual environment. This is also the oldest of the instances and was originally installed before venv came along
> 
> Ok that's quite old then? I saw this bit in your initial post, maybe you mean a different version.
> 
> > Octopi 0.17.0
> 

This is what is shown on the main screen below the files section on this one.
[image]
I have not installed OctoPi in a long time so I am guessing that running updates on the Pi would also update OctoPi.  Right?  I am assuming that if I originally installed OctoPi 0.15.0, the OctoPi version would be updated to reflect the updates.  
Also, on this one I can access Octoprint from a browser by using the IP address alone.  For example 192.168.100.1

The other two Pi's I installed Raspbian, and then Octoprint later.  One of them even has the DE installed as I was going to use it for something else but it didn't work out the way I wanted it to.  To access these from a browser I have to use the ip address:port method like 192.168.100.2:5000 and 192.168.100.3:5000

For all 3 of these, I run updates periodically, but not regularly.  Like every few months when I get a hair...nevermind.   I log into the Pi using ssh (Putty on Windows, or terminal/SSH command on Linux) and run "sudo apt update", "sudo apt upgrade", and "sudo apt dist-upgrade", and follow with "sudo reboot now".  It seems to me that this should have all 3 instances of Octoprint at the same level, but clearly something is different.

Just for curiosity, all three Octoprints show Python 2.7.16 at the bottom of the main screen.  Why doesn't it also show that Python 3 is installed???

jneilliii, 2020-08-30:
> Why doesn't it also show that Python 3 is installed???

OctoPrint only reports the version of the virtual environment it's running inside.

foosel, 2020-08-30:
> I have not installed OctoPi in a long time so I am guessing that running updates on the Pi would also update OctoPi. Right?

Wrong. OctoPrint only updates OctoPrint. `apt` only updates the underlying Raspbian. The only way for that OctoPi version number to change is by reflashing to a newer image (or manually fiddling around with the version file, but I'd consider that quite unlikely, especially by accident).

At some point you flashed 0.17 there.

xeddog, 2020-08-31:
Well, something is definitely not computing.  I haven't installed OctoPi in over 2 years and I don't think 0.17 was around then.  So as much as I hate to waste a day, I am going to rebuild all three Pi's.  Is there any way to install Raspbian, then Octoprint WITHOUT virtual environment?

jneilliii, 2020-08-31:
Why not just manually install the plugin that is not able to update automatically? I don't see that you ever did that.

cp2004, 2020-08-31:
I'm also not sure why you want it out of the virtual environment - that's not the cause of the issue here. Plugin is blacklisted so can't be updated - needs manual install. No more to it as far as I can see.

xeddog, 2020-08-31:
> I'm also not sure why you want it out of the virtual environment - that's not the cause of the issue here. Plugin is blacklisted so can't be updated - needs manual install. No more to it as far as I can see.

I don't want the virtual environment because this is a simple device that has only one simple task to perform on a simple and relatively secure network.  It doesn't need the added anti-simple, so KISS.

As for the plugin being blacklisted, the latest version is NOT blacklisted so it **should** update as it did on one instance.  The one that isn't in a venv.  Or at least I don't think it is.  Make that WAS.  I have re-Etcher'd all three copies with Buster, and I'm just about ready to start on OctoPrint.

Next, why not OctoPi?  Because.

foosel, 2020-08-31:
> As for the plugin being blacklisted, the latest version is NOT blacklisted so it should update 

It needs to be active for OctoPrint to even know where to update it from. It's blacklisted and thus not active. Reinstall via plugin manager, it's one button to click.

> I don't want the virtual environment because this is a simple device that has only one simple task to perform on a simple and relatively secure network. It doesn't need the added anti-simple, so KISS.

The venv makes it simpler as it avoids issues a system wide install has. System wide ain't supported as it complicates things and makes things easier to break.

The venv had nothing to do with your update issue, that's due to the blacklist as pointed out. Follow the solutions you've been provided with here or don't, that's your choice.

closed 2020-08-31 by foosel

xeddog, 2020-09-01:
I'm not going to pursue this anymore.  I have already re-imaged all three Pis (WITH venv dangit) and am now working on all of the customization (forgot to take a backup).  I still don't understand why one of three Pi's updated and two did not.  The plugin was blacklisted on all three before the one updated.  

Thanks all

xeddog, 2020-09-01:
After re-imaging all three of my Pi's, I came across a setting in the initial setup that I didn't even remember existed, and that was this:

[image]

I don't think I would have disabled this, at least not on purpose, but it would explain things if I had.  Anyway, I'm saying I screwed up and did, at some point for some reason, disabled it on two of them.  (long deep sigh)