OctoPrint/OctoPrint #3709: Help Octoprint stop working after install touchUi

VASAPOL, 2020-08-29:
I use Pi 3 with rpi display and The OctoPrint server is currently not running
Logs:
2020-08-28 15:17:47,154 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-08-28 15:17:47,241 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2020-08-28 15:17:47,242 - octoprint.server - INFO - OctoPrint 1.3.12
2020-08-28 15:17:47,248 - octoprint.plugin.core - INFO - 15 plugin(s) registered with the system:
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  Force Login (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/forcelogin
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Printer Safety Check (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/printer_safety_check
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
2020-08-28 15:17:47,464 - octoprint.environment - INFO - Detected environment is Python 2.7.16 under Linux (linux2). Details:
|  hardware:
|    cores: 4
|    freq: 1400.0
|    ram: 917016576
|  os:
|    id: linux
|    platform: linux2
|  plugins:
|    pi_support:
|      model: Raspberry Pi 3 Model B Plus Rev 1.3
|      octopi_version: 0.17.0
|  python:
|    pip: 19.3.1
|    version: 2.7.16
|    virtualenv: /home/pi/oprint
2020-08-28 15:17:47,465 - octoprint.server - INFO - ------------------------------------------------------------------------------
2020-08-28 15:18:01,605 - octoprint.server.preemptive_cache - INFO - ... done in 24.73s
2020-08-28 15:18:01,632 - tornado.access - WARNING - 404 GET /intermediary.gif?_=1598624255100 (::ffff:192.168.2.38) 16.05ms
2020-08-28 15:18:09,007 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.2.38
2020-08-28 15:18:23,697 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 56}
2020-08-28 15:18:24,420 - octoprint.plugins.tracking - INFO - Sent tracking event pong, payload: {'plugins': u''}
2020-08-28 15:18:25,196 - octoprint.plugins.tracking - INFO - Sent tracking event startup, payload: {'python': '2.7.16', 'ram': 917016576, 'octopi_version': '0.17.0', 'version': u'1.3.12', 'pip': '19.3.1', 'cores': 4, 'freq': 1400.0, 'pi_model': 'Raspberry Pi 3 Model B Plus Rev 1.3', 'os': 'linux'}
2020-08-28 15:18:42,678 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2020-08-28 15:18:47,833 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages/ (writable -> yes), --user flag needed -> no, virtual env -> yes
2020-08-28 15:18:47,834 - octoprint.util.pip - INFO - ==> pip ok -> yes
2020-08-28 15:18:47,848 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2020-08-28 15:18:52,490 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.2.38
2020-08-28 15:18:52,739 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.2.38
2020-08-28 15:18:52,843 - octoprint.server.util.flask - INFO - Passively logging in user dummy from ::ffff:192.168.2.38
2020-08-28 15:18:54,697 - octoprint.plugins.announcements - INFO - Loaded channel _important from https://octoprint.org/feeds/important.xml in 0.88s
2020-08-28 15:18:56,056 - octoprint.plugins.announcements - INFO - Loaded channel _releases from https://octoprint.org/feeds/releases.xml in 0.89s
2020-08-28 15:18:56,868 - octoprint.plugins.announcements - INFO - Loaded channel _blog from https://octoprint.org/feeds/octoblog.xml in 0.75s
2020-08-28 15:18:57,743 - octoprint.plugins.announcements - INFO - Loaded channel _plugins from https://plugins.octoprint.org/feed.xml in 0.82s
2020-08-28 15:18:58,710 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from https://octoprint.org/feeds/octopi.xml in 0.88s
2020-08-28 15:18:59,839 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin pluginmanager
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 68, in pluginData
    response = api_plugin.on_api_get(request)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 280, in on_api_get
    if refresh_repository or not self._is_repository_cache_valid():
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 797, in _is_repository_cache_valid
    return mtime + self._repository_cache_ttl >= time.time() > mtime
TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'
2020-08-28 15:18:59,853 - tornado.access - ERROR - 500 GET /api/plugin/pluginmanager (::ffff:192.168.2.38) 25.44ms
2020-08-28 15:19:38,424 - octoprint.server.api.system - INFO - Performing command for core:restart: sudo service octoprint restart
2020-08-28 15:19:38,700 - octoprint.server - INFO - Shutting down...
2020-08-28 15:19:39,131 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2020-08-28 15:19:39,134 - octoprint.events - INFO - Event loop shut down
[... 267 lines cut ...]
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1276, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1295, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_principal.py", line 199, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 307, in setSettings
    response = _saveSettings(data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/settings.py", line 505, in _saveSettings
    s.set(["temperature", "profiles"], result)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/settings.py", line 1658, in set
    self.load()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/settings.py", line 912, in load
    column=column)
InvalidSettings
2020-08-28 15:57:24,085 - octoprint.server - INFO - Shutting down...
2020-08-28 15:57:24,161 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2020-08-28 15:57:24,173 - octoprint.events - INFO - Event loop shut down
2020-08-28 15:57:24,197 - octoprint.server - INFO - Goodbye!

cp2004, 2020-08-29:
You're running OctoPrint 1.3.12, please try and update to the latest version - the issues/stack trace in your log may be fixed in newer releases. Its hard for us to investigate bugs on older releases, so please help us here.

If you're having issues with TouchUI please try the [community forum](https://community.octoprint.org) for support, this is the bug tracker for OctoPrint.

VASAPOL, 2020-08-29:
I already update all and issue found again.I reprogram 5 times with news version and olds version and issue found again.Please help

cp2004, 2020-08-29:
Can you provide a log of the newer versions please - again it's hard to look into old logs, since things change in the code.

VASAPOL, 2020-08-30:
My new logs @cp2004 
2020-08-30 05:52:05,566 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: u"Need(method='role', value=u'plugin_action_command_prompt_interact')")
2020-08-30 05:52:05,567 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: u"Need(method='role', value=u'plugin_announcements_read')")
2020-08-30 05:52:05,568 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: u"Need(method='role', value=u'plugin_announcements_manage'), Need(method='role', value=u'plugin_announcements_read')")
2020-08-30 05:52:05,570 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: u"Need(method='role', value=u'plugin_appkeys_admin')")
2020-08-30 05:52:05,571 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: u"Need(method='role', value=u'plugin_backup_access')")
2020-08-30 05:52:05,572 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: u"Need(method='role', value=u'plugin_firmware_check_display')")
2020-08-30 05:52:05,574 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: u"Need(method='role', value=u'plugin_logging_manage')")
2020-08-30 05:52:05,576 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: u"Need(method='role', value=u'plugin_pi_support_check')")
2020-08-30 05:52:05,577 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: u"Need(method='role', value=u'plugin_pluginmanager_manage')")
2020-08-30 05:52:05,579 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: u"Need(method='role', value=u'plugin_pluginmanager_install'), Need(method='role', value=u'plugin_pluginmanager_manage')")
2020-08-30 05:52:05,580 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: u"Need(method='role', value=u'plugin_softwareupdate_check')")
2020-08-30 05:52:05,581 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: u"Need(method='role', value=u'plugin_softwareupdate_update')")
2020-08-30 05:52:05,599 - octoprint.access.users - INFO - Migrating user Maker to new granular permission system
2020-08-30 05:52:05,639 - octoprint.plugins.softwareupdate - INFO - Version cache was created for another version of OctoPrint, not using it
2020-08-30 05:52:06,586 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2020-08-30 05:52:07,533 - octoprint.util.pip - INFO - Version of pip is 19.3.1
2020-08-30 05:52:07,535 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2020-08-30 05:52:07,536 - octoprint.util.pip - INFO - ==> pip ok -> yes
2020-08-30 05:52:07,542 - octoprint.plugin.core - INFO - Initialized 18 plugin implementation(s)
2020-08-30 05:52:07,832 - octoprint.plugin.core - INFO - 18 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/errortracking
|  File Check (2020.8.7) (bundled) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_file_check
|  Firmware Check (2020.6.22) (bundled) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_firmware_check
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/logging
|  Login UI (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/loginui
|  Pi Support Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
[... cut ...]
2020-08-30 06:18:56,254 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:49.230.15.2
2020-08-30 06:18:56,337 - octoprint.server - INFO - Shutting down...
2020-08-30 06:18:56,815 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2020-08-30 06:18:56,817 - octoprint.events - INFO - Event loop shut down
2020-08-30 06:18:56,818 - octoprint.server - INFO - Calling on_shutdown on plugins
2020-08-30 06:18:56,831 - octoprint.server - INFO - Goodbye!

cp2004, 2020-08-30:
No issue OctoPrint's end from that log. And if you've managed to update it, and install TouchUI and restart, then you must be into the server. 

If you need help with TouchUI, then get in touch either on the plugin's repo, or on the [OctoPrint Community Forum](https://community.octoprint.org), both of which are better suited for this quesiton.

closed 2020-08-30 by cp2004