OctoPrint/OctoPrint #3711: [Request] Plugin list dump

KenLucke, 2020-08-30:
**Is your feature request related to a problem? Please describe.**
Sometimes it is convenient to have a complete list of plugins and versions when not in the plugin manager (or, indeed even in OP) - especially useful for sending to plugin authors to look for conflicts.  While the plugin manager has the scrollable list, it's not in a form that can be used in this manner, as it cannot be copied.

**Describe the solution you'd like**
Please add to the Plugin Manager the ability to generate a list of currently installed plugins, with status, version, author/homepage link.  This could be either generated as a .csv or text file (for downloading), or as an .html web page opened in a new browser tab, which the user could view and/or print

**Describe alternatives you've considered**
none, really

**Additional context**
Add any other context or screenshots about the feature request here.

jneilliii, 2020-08-31:
http://<ip>/api/plugin/pluginmanager

Will return a list of all available plugins in the repo and all those installed on the machine and their status and additional details. The plugins key has all the installed plugins data and the repository key has all the available plugins and their data.

KenLucke, 2020-08-31:
Great, glad it's already there somehow.  Took me a second to figure out where to put that URL :) 

Thank you for all your work.

KenLucke, 2020-09-01:
Closing my own issue because the solution (in a way) already exists.

cp2004, 2020-09-01:
This info also comes through in the `octoprint.log` so if you are submitting an issue, then the information can be read from there.

I'm tempted to grab this and make feature, just a 'export list' button in the plugin manager. I know I could have used it when working out a bug conflict, to export them and then cross off one by one in a test.

Re-opened for now, since I think there may be more to this feature than already exists.

KenLucke, 2020-09-01:
> I'm tempted to grab this and make feature, just a 'export list' button in the plugin manager. I know I could have used it when working out a bug conflict, to export them and then cross off one by one in a test.

Exactly what I was asking for, I only closed it because there was a workaround, and didn't want to cause extra workload on the programmers
.

cp2004, 2020-09-01:
For me, workaround are not good enough 🙂 

When I first made my plugins, I was eating feature requests for breakfast. Its slowed down now, but I still (mostly) enjoy it.

Marking this as 'grabbed' since I'll take a look at doing it. If I can't, I'll report back here.

jneilliii, 2020-09-01:
There was a similar request on the community forum by Albert in regard to providing a separate endpoint that provides just the installed plugin's part and not the whole repo.

cp2004, 2020-09-05:
Thinking about this a little more, and wanted to get some opinions. What do you think of an import/export function, where you can export a list of plugins (covering this current FR), which could then be uploaded to a different instance of OctoPrint and would install all the plugins in the list.

Unfortunately that's beyond me I think, so maybe @foosel if you think this is a good idea it could go down as a feature request for that?

KenLucke, 2020-09-05:
> Thinking about this a little more, and wanted to get some opinions. What do you think of an import/export function, where you can export a list of plugins (covering this current FR), which could then be uploaded to a different instance of OctoPrint and would install all the plugins in the list.

This is basically what I do when I set up a new instance, by doing a backup of one, then doing a restore from that  backup to the instance that I want to clone, so that all plugins and settings are identical across identical machines.  I then simply change the things such as machine name, etc., that set them apart.

If you do this, it would be more convenient than doing a full install and then a restore as I do now.  You would need to include the option to either clone the settings, too, or do a fresh install of the plugins (or even a selectable list as to which option [clone , new or ignore] of install for each plugin.

foosel, 2020-09-07:
We could move the current plugin list backup functionality from the backup plugin into the pluginmanager and expose it on the UI. The backup plugin could then be adjusted to utilize it during backup/restore.

Regarding the setup of a new instance, that sounds like we'd need an additional "copy from backup" maybe, which would copy over all the settings but *not* the password salt and cookie hashs as it usually does. I'd consider that a different feature request though.

cp2004, 2020-09-07:
Adjusting the backup & restore plugin seems like a good idea, since the functionality already there. Reflecting on this, I've got a little bit too much to do at the moment to look into it and properly... so ungrabbing. 🙁

cp2004, 2020-11-26:
Pending acceptance, PR #3833 implements this feature. Can then in future be tied to the request to batch install, but that's more complex...

foosel, 2020-11-26:
PR merged, will be included in 1.6.0

commit d4af37ae (2020-11-26): :sparkles: Add plugin list export button

Closes #3711

Exports the same plugin list that the backup plugin uses, to the route `/plugin/pluginmanager/export`
closed 2021-04-27 by foosel