OctoPrint/OctoPrint #3714: OctoPrint seems to be causing the firmware to ignore endstops.

bholzer, 2020-09-01:
Having a very strange issue that I am struggling to troubleshoot.

I have OctoPrint running on a Pi, connected to my Ender 3 v2, which has Marlin 2 firmware.

If I home the printer while OctoPrint is disconnected from the printer, I have no issues at all.

However, when I have OctoPrint connected, homing the printer causes the carriage to run into the endstops and continue trying to move further. This happens whether I send the command from the printer or from the OctoPrint UI.

I have double and triple checked that all my print volume and bed settings are correct in OctoPrint and in Marlin.

bholzer, 2020-09-01:
After some more research, I have been able to troubleshoot a little further. I am sending `M119` while holding down my endstop switches, and only the y axis switch seems to be reporting as "triggered" when pressed. So, it seems that the switches endstops aren't being registered at all when OctoPrint/OctoPi is connected

foosel, 2020-09-01:
It's the responsibility of your printer and its firmware to properly handle endstops. There's no documented way for OctoPrint to even interfere with that. I'd a serial connection like the one OctoPrint creates is intended the culprit here it's a hardware or firmware bug and nothing that can be fixed on OctoPrint's side.

Feel free to seek support for this on the community forums, but an OctoPrint bug that can be acted upon this is not and thus the bug tracker is the wrong place.

closed 2020-09-01 by foosel

bholzer, 2020-09-01:
Just want to post my solution here in case anyone else runs into the same thing. Marlin had a change introduced very recently that broke things. Turns out the workaround is very simple: just keep an SD card in the printer

https://github.com/MarlinFirmware/Marlin/issues/19223