OctoPrint/OctoPrint #3715: [Request] Attach gcode to pause/cancel/resume buttons

KenLucke, 2020-09-01:
**Is your feature request related to a problem? Please describe.**
When I cancel a print, I like to raise the nozzle (and frequently x/y home the head) so that it doesn't cool while still in contact with the last bit printed.

**Describe the solution you'd like**
The ability in the settings to add a gcode scrippet to those buttons.  This would allow you to send a raise Z and/or Home X/Y command frooom wherever those buttons appear, without having to go to the Control Tab to do it manually.  Pause could raise the Z your set amount, Resume could lower it that same amount, all automatically.

**Describe alternatives you've considered**
The only alternative is to go to the Control tab,  or use the "Macro" plugin or "Terminal Commands Extended" (or similar) buttons, and do it manually every time.

**Additional context**
Add any other context or screenshots about the feature request here.

cp2004, 2020-09-01:
Forgive me if I'm missing the point of your request, but is this not already possible through G Code scripts in OctoPrint?
You'd be able to put them here:
[image]

KenLucke, 2020-09-01:
Damn, stupid me, completely forgot about that,  Sorry

closed 2020-09-01 by KenLucke