OctoPrint/OctoPrint #3717: allowCrossOrigin=true but API still only reachable on localhost

OevreFlataeker, 2020-09-03:
#### What were you doing?

Trying to expose the API to hosts outside of octoprint system. The option "Allow Cross Origin Resource Sharing" has been enabled and Octoprint has been restarted.

The config file shows the value set to true:
```
pi@octopi:~/.octoprint $ head config.yaml 
accessControl:
    salt: xxxx
api:
    allowCrossOrigin: true
    key: xxxx
plugins:
    _disabled:
    - touchui
    - SpoolManager
    - kronos_data_collector
...
``` 
however the port for the API is still bound to localhost after a restart of octoprint and is thus not available to the outside
```
pi@octopi:~/.octoprint $ netstat -anpA inet | grep :5000
(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
tcp        0      0 127.0.0.1:5000          0.0.0.0:*               LISTEN      7471/python2        
tcp        0      0 127.0.0.1:60130         127.0.0.1:5000          ESTABLISHED -                   
tcp        0      0 127.0.0.1:5000          127.0.0.1:60130         ESTABLISHED 7471/python2        
```
#### What did you expect to happen?

API will listen on 0.0.0.0:5000
#### What happened instead?

API still listens on 127.0.0.1:5000

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.4.2 on octopi 0.15.1

#### Operating System running OctoPrint

OctoPi 0.15.1

#### Printer model & used firmware incl. version

Anycubic i3 Mega

#### Browser and version of browser, operating system running browser

Chrome, Linux/Windows

#### Link to octoprint.log

https://pastebin.com/Q1h6K4xu

#### Link to contents of terminal tab or serial.log

https://pastebin.com/dFEMm3Hu

#### Link to contents of Javascript console in the browser

n/a

#### Screenshot(s)/video(s) showing the problem:

n/a

I have read the FAQ.

jneilliii, 2020-09-03:
This potentially is because of how the octopi image is crafted to use haproxy? What happens if you just use the api endpoint htt[://192.168.0.2/api/... replacing with your IP address of course.

foosel, 2020-09-04:
As @jneilliii already indicated, the bound address has nothing to do with the CORS setting (and if it did that would be really weird as one has nothing to do with the other from a conceptual point of view). 

On OctoPi OctoPrint gets started with `--host 127.0.0.1` as there's an instance of haproxy in front, making everything available on the http standard port 80. Hence there's no need for OctoPrint itself to bind to any other address but localhost. 

Use port 80 instead of port 5000, or edit `/etc/default/octoprint` to change the host bind (though why you'd want to do that is beyond me, going through haproxy should really be the preferred and default way to go here and had no disadvantages at all).

closed 2020-09-04 by foosel

OevreFlataeker, 2020-09-04:
Hm, so port 5000 should not be exposed at all? I was wondering about that CORS setting as well, as it did not seem to be directly related to the listening port setting but that was the best I found googling for my issue tbh. I can confirm access to e.g. http://octopi/api/connection works as intended. I assumed I HAD to go via port 5000. Thank you.