OctoPrint/OctoPrint #3720: Backup always includes all uploaded files

spanzetta, 2020-09-09:
Even if I click/flag the option  **"Exclude uploaded files & metadata from backup"** when I create a Backup it always includes also the uploaded Gcode files.. creating a large Backup file..

Pls check

Raspberry PI 3A+
OctoPrint 1.4.2 
Python 2.7.16 
OctoPi 0.17.0

spanzetta, 2020-09-10:
anyone can check and confirm?

foosel, 2020-09-10:
I cannot confirm, works as designed over here. You were also asked by the bot to fully fill out the ticket template, provide logs etc. Please do so, otherwise this cannot be looked at.

spanzetta, 2020-09-10:
hi @foosel 
thanks for the reply..
I've read the "automatic" answer... and frankly speaking I think that relevant information (versions of each component) waS already included ..  

Pls note that it happens always on my setup..

I now add the octopring.log which I didn't include before (my fault)

Just let me know if some other relevant information or log is missing.. and I will provide.

Waiting some feedback 
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5201590/octoprint.log)

Thanks in advance

foosel, 2020-09-10:
> and frankly speaking I think that relevant information (versions of each component) was already included .. even if without a "template" (does it really matter?)

Yes it does really matter. I still don't know if it did the same in safe mode, as you already noticed yourself you didn't include the log initially which like **the** most important thing to include as it tells me a lot about your environment even if it tells me nothing about the bug itself.

Please in the future just fill out the template, it is there for a reason. I don't want and also can't constantly run after info needed to help people, and trying to parse out the required information from a wall of text again and again while trying to debug the issue also eats up a ton of time that could be better spent on actually solving issues. It's not much overhead on your side and means I have more time to help you and everyone really. Just be considerate and play by the rules, this job is stressful enough ;)

----

Now, concerning this ticket, please do this:

  1. Try if the issue persists in safe mode, report back whether it does or doesn't.
  2. If it does please share a backup with me *privately* (do not share it here publicly, there might be sensitive information in there!). Upload it somewhere and then send me the link in a mail to gina@octoprint.org. I'd like to take a look into what gets generated and if it's actually the uploads that get included or something else maybe that blow up the size.
  3. Test if excluding both timelapses *and* uploaded files makes a difference in file size.

I haven't been able to reproduce the behaviour, when I exclude uploads they get excluded.

spanzetta, 2020-09-10:
Update: 
I noticed that Gcode files are incluede in the ZIP under the folder \basedir\data\UltimakerFormatPackage when the option is enabled..
If the option is disabled then GCode are stored 2 times .. under \basedir\data\UltimakerFormatPackage and also under \basedir\uploads 
I hope this helps ..

I will follow your suggested points (1 to 3) shortly..

Thanks for the patience.. ;-)

jneilliii, 2020-09-10:
I thought I had made the plugin clear those gcode files once successfully uploaded,  but I suppose I didn't, wil have to go back and review. But yeah, it would make sense to expose something like `self.disable_plugin_data_backup` or something like that because if you're not backing up gcode then there's no reason to back up thumbnails either, and this would apply to PrusaSlicerThumbnails as well.

spanzetta, 2020-09-10:
at the end we (you) found the root cause.. ;-)
great!
@jneilliii let me know if I need to open the ticket on your page or if you will manage it anyway..
Thanks for both Gina (always great!) and @jneilliii

jneilliii, 2020-09-10:
Right now, the only thing I would be able to eliminate is the gcode files I think and the thumbnails will still get backed up until there is some way of flagging those exclusions in the backup plugin in the future once that's available. WOuld be helpful if you could create an issue on my repo though.

spanzetta, 2020-09-10:
great! So fast!
Thank you so much...

closed 2020-09-10 by spanzetta

jneilliii, 2020-09-10:
Thanks @foosel I'll have that updated in my next release of the plugin.