OctoPrint/OctoPrint #3726: Upload to SD fails in Python 3 environment

b-morgan, 2020-09-12:
#### What were you doing?

1. Connect to printer (LulzBot TAZ 6).
2. Click on 'Upload to SD'.
3. Pick a file (I used a small .gcode file that had successfully printed. I think any file will do).
4. See error

#### What did you expect to happen?

The file to be transferred to the printer's SD card 

#### What happened instead?

2020-09-11 19:49:30,221 - See octoprint.log for details
2020-09-11 19:49:30,258 - Changing monitoring state from "Sending file to SD" to "Offline (Error: See octoprint.log for details)"
2020-09-11 19:49:30,261 - Connection closed, closing down monitor

2020-09-11 19:49:30,084 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 2185, in _monitor
    self._changeState(self.STATE_PRINTING)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 697, in _changeState
    self._callback.on_comm_state_change(newState)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/printer/standard.py", line 1116, in on_comm_state_change
    self._analysisQueue.pause() # only analyse files while idle
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/analysis.py", line 106, in pause
    q.pause()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/filemanager/analysis.py", line 218, in pause
    self._do_abort()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 207, in _do_abort
    if _allow_analysis(self._plugin._printer, self._plugin._settings):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 188, in _allow_analysis
    if thermostat['target'] < 30:
TypeError: '<' not supported between instances of 'NoneType' and 'int'
2020-09-11 19:49:30,258 - octoprint.util.comm - INFO - Changing monitoring state from "Sending file to SD" to "Offline (Error: See octoprint.log for details)"
2020-09-11 19:49:30,296 - octoprint.plugins.action_command_notification - INFO - Notifications cleared
2020-09-11 19:49:30,843 - octoprint.plugins.tracking - INFO - Sent tracking event commerror_crash, payload: {'commerror_text': 'See octoprint.log for details'}
2020-09-11 19:49:30,850 - octoprint.filemanager.analysis - INFO - Analysis of entry local:Test_Cube.gcode finished, needed 0.97s
2020-09-11 19:49:46,021 - octoprint.server.api.system - INFO - Performing command for core:restart_safe: sudo service octoprint restart

#### Did the same happen when running OctoPrint in safe mode?

No, but the error above compels me to enter the bug anyway.

#### Version of OctoPrint

1.4.2

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

LulzBot TAZ 6 w/ M175 tool head, Marlin 2.0.0.144

#### Browser and version of browser, operating system running browser

Firefox 80.0.1 (64-bit)

#### Link to octoprint.log

https://gist.github.com/b-morgan/b5c875e48992343f258c81965fdf0046

#### Link to contents of terminal tab or serial.log

https://gist.github.com/b-morgan/b5c875e48992343f258c81965fdf0046

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

N/A

Yes, I have read the FAQ.

b-morgan, 2020-09-12:
Also reported in [https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/210#issue-699977451](url)

Grogyan, 2020-09-12:
Upload to SD used to work just fine on Python2

Ver 1.4.2
OctoPI 0.17
Marlin 2.0.5.3

Under Python 3, once a gcode is uploaded to SD, OctoPrint closes the serial connection to the printer, and attempts to upload to a closed serial port.

Plugins disabled
Still stays stuck in upload
Serial log snippet
020-09-11 00:44:45,900 - Send: N3851 M105*24
2020-09-11 00:44:45,913 - Recv: Error:error writing to file
2020-09-11 00:44:45,916 - Recv: ok
2020-09-11 00:44:46,694 - Recv:  T:19.25 /0.00 B:18.12 /0.00 @:0 B@:0
2020-09-11 00:44:48,694 - Recv:  T:19.25 /0.00 B:17.97 /0.00 @:0 B@:0
2020-09-11 00:44:50,693 - Recv:  T:19.25 /0.00 B:17.97 /0.00 @:0 B@:0
2020-09-11 00:44:50,902 - Send: N3852 M105*27
2020-09-11 00:44:50,905 - Recv: Error:error writing to file
2020-09-11 00:44:50,907 - Recv: ok
2020-09-11 00:44:52,692 - Recv:  T:19.25 /0.00 B:17.97 /0.00 @:0 B@:0
2020-09-11 00:44:54,692 - Recv:  T:19.25 /0.00 B:17.97 /0.00 @:0 B@:0
2020-09-11 00:44:55,902 - Send: N3853 M105*26
2020-09-11 00:44:55,911 - Recv: Error:error writing to file
2020-09-11 00:44:55,923 - Recv: ok
2020-09-11 00:44:56,693 - Recv:  T:19.25 /0.00 B:17.97 /0.00 @:0 B@:0

Full logs on forum

cp2004, 2020-09-12:
So from @b-morgan's logs, this is a Print Time Genius issue: it fails to analyse properly and then the SD transfer is aborted. I see you opened an issue there, and as you tested it with safe mode and it DID NOT occur, I can be confident saying this is not an OctoPrint bug.

For @Grogyan - your issue does appear to happen with actually transferring the files across. Could you confirm that the same thing happened in safe mode?

Grogyan, 2020-09-12:
@cp2004 as there are some plugins which are absolutely necessary, all irrelevant plugins such as bed visualizer have been disabled. Which is as close to stock mode as I can get away with.

Problem still continues

cp2004, 2020-09-13:
@Grogyan I'm going to ask you to create a new bug report, with the full template, since your issue is different than @b-morgan's which appears to be with PTG. Otherwise we are getting confused with what the *actual* problem is, and what information is required.

Please provide all of the information requested in the template, to help this get analysed properly. It would be good if (in the new report) you could add some info with exactly which plugins you have, as well as the full logs **(incl. `serial.log`)

Grogyan, 2020-09-13:
@cp2004 full logs are on this forum post
https://community.octoprint.org/t/octoprint-python3/24008/10

Please review, then advise if it is appropriate for another issue to be created
Not ruling out that it may be a Print Time Genius bug as well

cp2004, 2020-09-13:
That's what safe mode would probably rule out... What plugins do you need that mean safe mode is not an option?

b-morgan, 2020-09-13:
@Grogyan If you want this issue fixed, you should cooperate with the developers. Open a new issue and fill out the template completely including testing in safe mode. Anything less and you are just wasting people's time.

Grogyan, 2020-09-13:
@b-morgan I specifically didn't raise a ticket, and instead raised on the forum to see if anyone had a similar issue.

Saw that you had a similar problem then forthwith raised tickets.  As they could be manifested differently for the same issue, thus added further details here.

I was not able to do any further testing in the weekend due to needing to actual printing, as I couldn't do that from SD, and had to gamble that there wasn't going to be an issue with the USB comms. USB is too susceptible to EMI for printing with.  When uploading to SD the USB comms have low EMI, and had worked for years under Py2 environment.
A 13 hour print was wasted, gamble lost

I hope to do more formal testing tonight, under safe mode, and with more plugins disabled.
Temp Sidebar
Print Time Genius
AstroPrint
Which are essential for any print over half an hour.

foosel, 2020-09-14:
They are not essential for verifying whether this is a 3rd party plugin issue or not. It doesn't make sense to look into this from my end until you have verified that it occurs in safe mode as well (which disables all 3rd party plugins). If it doesn't happen in safe mode it is a third party plugin and thus not a bug that I can fix. Of it does happen in safe mode, it is a different issue than the one here (and something I *can* fix). It's as simple as that.

I'm going to reclassify this one here as an improvement, since the only thing I can do here is more error handling to capture misbehaving third party plugins.

Grogyan, 2020-09-14:
@foosel I managed to have some time last evening to do a deeper dive.
Under Safe Mode
-  Uploading gcode to the SD card works.

Disabling each plugin in turn
-  Print Time Genius Plugin was causing the issue
[https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/210](Print Time Genius Py3)

Have updated @b-morgan issue for that plugin

When I reviewed older logs when my OctoPrint was under the Python 2 environment, the same bug was there for the plugin, however,  OctoPrint 1.4.2 and earlier carried on behaving as expected when upload gcode to the printer SD card.

Further logging may help the devs for that plugin to identify the root issue
@cp2004 so different symptom for the same issue.  No need to create another OctoPrint issue, but I feel this issue should probably be a Feature Request for more logging if possible, to report exceptions for plugin devs.

foosel, 2020-09-15:
Added a try catch block to where the (possibly third part) analysis queue gets called from the comm layer. That should reduce the disruptiveness of bugs like that starting with OctoPrint 1.5.0. Added my analysis to the PrintTimeGenius ticket about the nature of the bug.

commit cdce9a6f (2020-09-15): 🥅 Better resilience against broken analysis queues

Closes #3726
closed 2020-11-30 by foosel