OctoPrint/OctoPrint #3728: [Request] Cloud backup (on Google Drive?)

spanzetta, 2020-09-13:
In order to save time in case of Octoprint (Octopi) reinstall when SD breaks (it did happen to me) it would be useful to add a feature to Octoprint to make backup not only locally but also in Cloud.

I personally think the most usefull (and common) would be on Google Drive, but it could be done also on other Cloud service.

During Octoprint/Octopi setup wizard it could be added the feature to ask if you have a backup saved on Google Drive for an automatic Restore process .. for a very easy reinstall

jneilliii, 2020-09-13:
You could achieve this through scripting and a cron job. The backup command can be run from the command line, just don't know if you can specify the location.

spanzetta, 2020-09-14:
Yes.. it could be done even manually .. but an integrated feature will make the job much easier and smooth for a seamless user experience

spanzetta, 2020-09-14:
Just to be more clear:
What I am suggesting/proposing is a backup feature like the one included in WhatsApp: automatic/timed backup on Google Drive providing Google credentials (secured trough Google API/Oauth)..

Also the restore process at initial/new Setup could be done by simply asking to put Google credentials to check for Cloud backups; this operation of course could be skipped if someone doesn't have Cloud backups

Also Cura has a sort of Cloud backup.. I'll like to see it also on OctoPrint 

;-)

Thanks for the attention

foosel, 2020-09-16:
Problem with something like this is getting the auth to work in the first place. OAuth as implemented by Google's APIs isn't really tailored towards "service to be trusted is an arbitrary endpoint on a user's LAN". 

Additionally, the second I implement something like support for one cloud provider, I'll immediately get requests for further and suddenly I have not only a print server to maintain but also a whole host of cloud integrations just for backup uploading. So... what I'll do here instead is add an event on backup creation, and then someone can write a plugin for Google Drive/for AWS/for NextCloud/for whatnot to sync backups with their preferred cloud provider of choice.

spanzetta, 2020-09-16:
Ok.. it looks a good compromise..

One question.. with the plugin approach it will impossibile to integrate the restore functionality during setup..  isn't it?

Thanks

foosel, 2020-09-16:
Yes, but frankly having to download your backup file from your cloud storage before uploading it in that dialog will be way way easier and less of a headache for everyone than trying to even begin to figure out how to get access to cloud storage from a freshly setup instance somewhere on someones LAN, regardless of whether that would be from a plugin or core, and having that fail half the time due to network issues, invalid/not on hand credentials or having to jump through the hoops of having to do a device oauth workflow ("go to this page, you'll see a pin, enter it in this input field") which is the only thing possible with arbitrary endpoints on user LANs anyhow.

spanzetta, 2020-09-16:
Ok.. I was just suggesting exactly what Whatsapp does from a while... 
My idea was to make things very easy for customers .. that's it..
Thanks anyway..

foosel, 2020-09-16:
No worries, I get that. The problem is, OctoPrint is not Whatsapp. Whatsapp has a central server on the net, OctoPrint runs in your LAN. That makes everything involving cloud servers that require authentication difficult (at least as long as they use OAuth with authenticated endpoint URLs, which Google and several more of the big ones do). Whatsapp also has more than one developer 👋 and way better funding ;)

foosel, 2020-09-16:
Just pushed a commit that adds the event on backup creation, will be part of 1.5.0

jneilliii, 2020-09-16:
I like this approach and may eventually look into creating a google drive backup plugin to utilize this new event hook if @spanzetta  doesn't beat me to it.

spanzetta, 2020-09-16:
> I like this approach and may eventually look into creating a google drive backup plugin to utilize this new event hook if @spanzetta doesn't beat me to it.

Dear @jneilliii 
thanks for your effort.. I will be very happy to test it when it's ready..
Regards

jneilliii, 2020-09-16:
I've got the plugin saving to my Google Drive, just have to get the front end components created to be able to generate an oAuth token. It's a little tricky setting up initially because of having to create an app in your Google Account (similar to YouTube Timelapse uploads) but fairly easy to follow instructions for doing that.

jneilliii, 2020-09-17:
UI for authorizing Google Drive is now working, just need to document the steps for creating your own custom app and authorizing it within the plugin. Once the documentation is completed I'll give you an update.

spanzetta, 2020-09-18:
Thanks a lot for the update

jneilliii, 2020-09-19:
So I got the plugin documented and have someone testing it for me over the weekend. Here it is for reference, but still won't work unless you switch OctoPrint to install from the `github commit` and point it to the maintenance branch. 

https://github.com/jneilliii/OctoPrint-GoogleDriveBackup

I created a companion scheduling plugin for #3732 that should work with current OctoPrint stable release version.

https://github.com/jneilliii/OctoPrint-BackupScheduler

spanzetta, 2020-09-20:
Dear @jneilliii 
first of all thanks for the effort/job.
I saw the described procedure (I know it from another sw that was using Google API authentication).. but frankly speaking it is quite complex for the majority of users..
Wasn't possible to make a "hidden" script that does all the job .. so the user just need to put it's credentials (google email and password) and click on "authenticate"?
One sw that I know that works in this way is GSyncit (so it means that it's possible to do that way)  .. if you want to take a look on it here is the link
https://www.fieldstonsoftware.com/software/gsyncit5/
Let me know what do you think about
Thanks

jneilliii, 2020-09-20:
I'm still actively developing this plugin, and since it's still a little cumbersome to achieve the OAuth with a personal developer app that is the only way that this plugin will remain free and open source. I'm not going to pay for the stuff required to create a shared Google App and go through the verification process, etc. That being said, I am working on simplifying the process.

spanzetta, 2020-09-20:
Ok.. thanks for the update 
If paying to Google is the only way to make the process easy i agree with you that it's not a good way..
On the other side, if the process is too complex.. only few expert users will use it.. and that will be a pity
I will check your progress
Thanks a lot

foosel, 2020-09-20:
And now you know why I said this isn't as easy to pull off as WhatsApp and friends make it out to be.

spanzetta, 2020-09-20:
I was assuming that Google stuff.. since it's so widely used .. was also "easy" to program..
Instead it appears to be not so easy..
But it's also a challenge be able to do not only "easy things".. but also "complex things"..
Isn't it?
;-)

jneilliii, 2020-09-20:
It's not that it's complex, I completed the process in roughly a day's worth of work. It's that with an open source application you cannot include the files required to authenticate the access, as it is a secrets file. It would be possible to obfuscate that file and deliver it along with the plugin, but that can be reverse engineered because you have access to the source. 

Even if I were to do that, Google limits the number of requests to the API for free based on app, user, etc. and any usage over those limits would be charged. I could of course do that, but then I'd have to recoup that money somehow, by providing a secrets file via Patron subscription tier level possibly to pay for the API usage, the hosted server that would be required for redirects, etc. 

The documentation is detailed for that reason and may seem daunting, but it is easy to follow step by step and believe it to be a good compromise over forcing people to pay.

spanzetta, 2020-09-20:
thanks for the detailed explanation...
happy working..

spanzetta, 2020-09-23:
Hi @jneilliii 
I was trying to install the plugin but it says it's incompatible.. as I don't have Octoprint 1.5 (not released yet)..
[image]

jneilliii, 2020-09-23:
That is true. The event hook that the plugin uses to do the work isn't available until octoprint 1.5.0.

spanzetta, 2020-09-24:
ok.. so need to wait its release to test it..
thanks

foosel, 2020-11-30:
1.5.0 has just been released.

closed 2020-11-30 by foosel