OctoPrint/OctoPrint #3729: Possible CPU race condition and memory leak in Chromium based Browsers

psonoda, 2020-09-13:
#### What were you doing?
I was enabling and disabling plugin and experienced higher than normal CPU usage with my Chrome and Edge browser afterwords. The only workaround was to close the browser tab or to use FireFox instead of Chrome or Edge.

1) Open Task Manager within W10 to monitor CPU usage of your Octoprint browser tab
2) Go to manage plugins
3) Quickly enable or disable a single plugin (using a non-essential plugins make this easier)
4) Watch the task manager while you're repeating step #3 after 3-30 cycles your CPU will increase and stay. You will also see an increase in memory for each cycle.
5) The only way to clear the browser CPU usage and memory is to close the browser tab

#### What did you expect to happen?
Have the CPU usage in the browser return to it's normal 1-5% and memory usage return to normal

#### What happened instead?
Browser CPU and memory usage stayed high

#### Did the same happen when running OctoPrint in safe mode?
Yes, tried in safe mode with no printer or webcam. It seems to be a little harder to duplicate but I have duplicated it many times in safe mode, no printer.

#### Version of OctoPrint
OctoPrint 1.4.2

#### Operating System running OctoPrint
3B+
Python 2.7.16
OctoPi 0.17.0

#### Printer model & used firmware incl. version
MK3S but issue happens with no printer attached (off)

#### Browser and version of browser, operating system running browser
W10 in Chrome 85.0.4183.102 (Official Build) (64-bit) and Edge Version 85.0.564.51 (Official build) (64-bit)

#### Link to octoprint.log

https://gist.github.com/psonoda/f80033593ba004ed6d4b7d6386c9800d
No associated log entries are created during the test

#### Link to contents of terminal tab or serial.log
Happens with PRINTER off

#### Link to contents of Javascript console in the browser
https://gist.github.com/psonoda/e758f7b966f0a6db2bd169b08c4f2e87

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2020-09-15:
I can confirm this under Chrome. I can also confirm that it doesn't seem to happen under Firefox. 

First analysis hints at an accumulation of a lot of requests seems to be at fault, caused by constant refreshing of the plugin list after toggling a plugin. I first suspected the generated notifications to be at fault here, but after I reduced it to one that's being reused I still could reproduce, so that's most likely not it (or the reuse is the problem). I've now made sure that only one refresh request can ever be active at once, though I'm not 100% happy with that yet (as we might get somewhat stale data if someone clicks really fast). I still see an increase in memory and CPU, but that seems to normalize after a while (my guess is after the next GC maybe?). Poking at it with all the profiling tools in the developer tools so far hasn't unearthed anything specific.

@psonoda could you maybe test if you can still reproduce on the current `maintenance` branch?

psonoda, 2020-09-15:
Got the dev version installed. I can still duplicate the issue but it is a LOT harder. A lot more rapid clicks. I am also still leaking ram but it is factor of 10x or more less. I really like the new behavor of the message screen. I found the previous bouncing of the screen annoying.
[image]
[image]

foosel, 2020-09-16:
Some more de-concurrency changes as those seem to cause these issues. 

I'm still seeing increased memory usage on multiple toggling, however the actual heap stays small in the 30MB range. For some reason Chrome reserves a ton more memory for the tab once memory usage increases temporarily and won't give it back even on forced garbage collects. Starting the memory profiler however makes things go poof. This behaviour frankly seems to be outside of my control and I'd note it up to the general browser implementation ("memory is cheap"). I don't see increased CPU usage anymore, it stays up for a bit, possibly while some timers run out, and then it returns to 0-0.5% while idle. 

[image]

[image]

[image]

psonoda, 2020-09-16:
Hi Gina, well I can still duplicate but it's even harder. It seems to be much easier to duplicate in Edge over Chrome. A lot of fast clicking can duplicate the issue. Also I've noticed a new behavor in the messaging system. If you click on the "X" and close the message you no longer get any messages until you reload the page.

foosel, 2020-09-21:
> If you click on the "X" and close the message you no longer get any messages until you reload the page.

That was a bug introduced by switching to only having one reused notification. I've fixed that in cac1e4b0bd78bd6b4c0e5ff394b960a67f6d3e7f.

Other than that, I really am not able to reproduce anything that does look like it's actually an issue with the JS app. I just fired up Edge and took a look at heap size and task manager there as well and it looks pretty much identical to Chrome.

[image]

Do you still see increased CPU usage or is it just the memory consumption reported by the **OS** task manager when you reproduce? Also, a lot of fast clicking on what? The enable toggle? Because that will now block while a request is in process, no matter how fast I click does cause any kind of request cascade.

psonoda, 2020-09-21:
I was still able to get the high CPU but it wasn't easy. But let me give it another try with your latest version. I will send screen shots. I've been up all night working on a screw ball Windows Server work issue so it may be a day or 2.

psonoda, 2020-09-22:
I installed cace41b. I was still able to duplicate the issue. See screen shot. The way I do my test is pretty straight forward. 1) Make sure I am not running in OP safe mode. 2) Go into the plugin manager and find a non-essential plugin. 3) Turn it on/off by clicking on it as fast as I can. I don't see excessive memory use just the CPU loading. I can confirm that the error message does now reapear after closing without reloading the page. Also, I did notice I couldn't get OP to restart from the main menu. This is new behavor for me on the maintenance version. It would give me the restart message but never actually restart. I had to restart OP from the ssh command prompt.

[image]

foosel, 2020-09-22:
Cannot reproduce. CPU briefly goes up a bit during click storm but then returns to 0 upon end of generated requests. I'll mark this as unreproduced for now since I'm no longer able to see what you are seeing with current `maintenance`. I frankly cannot justify spending more time on this right now considering that it only reproduces if it even reproduces at all via excessive clicking on a toggle which usually will not be excessively clicked on - limited development resources. If anyone else wants to further investigate this and finds something, that would be lovely.

About the other thing, I didn't touch the restart stuff in any way, and if you couldn't restart it through the system menu either something must have become corrupt in your config, or the way you started OctoPrint's `maintenance` branch didn't match the configured restart command (e.g. if you start it manually via SSH via `octoprint serve`, `service octoprint restart` won't do a thing).

psonoda, 2020-09-22:
As a final follow up, I did a clean install. This issue should not happen in "real life" anymore. From what I am seeing is that depending on the speed of the computer/browser and the speed of clicking it is potentially possible to make this issue happen. But no one should be doing this IRL. I am still unable to restart OP from the main menu or from a restart message button. It shows a restart on the screen but OP server doesn't actually restarts. I understand that you haven't touched the code but I did a clean install from scratch of the files and the venv and it's still not restarting (FYI I am still on Python 2.7 due to a plugin). I've tried starting the server with "venv/bin/octoprint server" and "venv/bin/octoprint daemon". Cheers, Paul

foosel, 2020-11-30:
1.5.0 has just been released.

closed 2020-11-30 by foosel