OctoPrint/OctoPrint #3732: [Request] Automatic backup by backup plugin

hamster65, 2020-09-16:
Octoprint's build-in backup feature is great and always worked for me, but you have to remember to do the backups.

This can be automated via command line, but wouldn't it be nice to have an option directly in the plugin for daily, weekly, monthly backups, or a backup on startup of Octoprint and the path where to save the backups, and how many backups to keep?

Copying those backups to a remote system should be left to the user.

jneilliii, 2020-09-19:
I've made you a plugin @hamster65 if you want to test it out before I submitted it to the repo.

https://github.com/jneilliii/OctoPrint-BackupScheduler

hamster65, 2020-09-19:
Awesome, thanks! :)

closed 2020-09-19 by hamster65