OctoPrint/OctoPrint #3733: [Request] Permanent ignore of a particular update notification

cp2004, 2020-09-16:
**Is your feature request related to a problem? Please describe.**
A slightly annoying thing for me, when developing plugins, is the update notifications keeps coming back for my plugins installed in editable mode with pip - and as a result the version number is not properly incremented. Every time a new update is available, the notification for my plugin shows up as well.

As a result, clicking the 'update now'  button is not an option, since it would overwrite my editable deployment.

**Describe the solution you'd like**
Permanent ignore for the notifications. I thought maybe it could be turn off notifications for that plugin, but then I thought users might disable notifications for plugins and not update *ever*.

So the solution I thought of was to ignore that specific update notification, so that you still get notified when there is a new version, to get people to update still, if they ignore it.

**Describe alternatives you've considered**
Running `pip install -e .` or `octoprint:dev plugin install` each time I increment the version number.

**Additional context**
If it helps, here's a screenshot of the problem
[image]

I would like to update OctoPrint, but not my plugin since it would overwrite editable mode.

Hopefully that makes sense, if you need further clarification I'm always here....

jneilliii, 2020-09-20:
I wonder if it would make more sense to have an option to disable software update checks while running in debug mode maybe? Not even sure if that's possible.

cp2004, 2020-09-20:
I started poking at the code to see if this was more like a bug (when you press ignore I would *expect* it to not come back) but still undecided.

foosel, 2020-09-21:
I just checked, and it's not a bug but intentional. The idea is to only be able to ignore that specific combination of updated versions. So if for example you have Plugin A at offered version 1.0.0 and Plugin B at 0.3.2, you can ignore it, but the notification for both will show up again if either Plugin A or Plugin B push out a new version.

But I think being able to completely disable update checks would be a good idea in general, so I'm going to add that.

cp2004, 2020-09-21:
That does kind of make sense, I probably pressed ignore when it was bundled with something, I will try and see if it doesn't come back if it is on its own.

foosel, 2020-09-21:
Bell in first column will now be muted if no update is available and can be used to disable update notifications for the entry in question, in which case a slashed bell will be used as the notification icon:

[image]

Clicking on "Update Now" in the notification

[image]

Clicking on "Update All" in the settings:

[image]

cp2004, 2020-09-21:
🎉 Looks good

foosel, 2020-11-30:
1.5.0 has just been released.

closed 2020-11-30 by foosel