OctoPrint/OctoPrint #3734: Possible issue on print time

Techrese, 2020-09-18:
Printer: ultimaker original + 
using latest version cura engine

see screenshot:
[image]

using cura engine print time 2hours
octoprint approx 2hours
however print time left is like 9 hours.
Is this normal or is this a possible bug as the print time left is 2 times more that the actual print time.

I assume the print time left is based on where the gcode is?

foosel, 2020-09-18:
> Is this normal or is this a possible bug as the print time left is 2 times more that the actual print time.

Well documented limitation of how print time is calculated, discussed ad nauseam in OctoPrint On Air, the forums, various tickets etc etc. Linear approximation of a non linear process can have such side effects. See the red bubble next to the estimate telling you it's low accuracy. 

And for questions in the future please first use the [community forums](https://community.octoprint.org) ;)

closed 2020-09-18 by foosel

Techrese, 2020-09-18:
ow , oke sorry to bother then, but thanks for the information