OctoPrint/OctoPrint #3737: Loading over 100%

whoim2, 2020-09-20:
https://prnt.sc/ukkigl

I'm tired. For one year I have been trying to get octoprint to work without interruptions that cause material to bleed when printing. I am using RPi3B (I tried RPi3A and RPi3B + before), the latest version of windowprint on octopi, clipper on skr e3 mini.
There is also a display and octodash, but when they were not there, there were also stops.
I added a fan and this lowered the maximum temperature from 80 to 55 degrees. I always put radiators.
I increased the buffer to 15 units in the config. The day before, I enabled force_turbo = 1 and printed several models without freezing. I was sure I had won.

Today I set the model to print, and 95% of it was printed without hang-ups, but the last 5% were accompanied by two or three freezes per layer. And it practically ruined the whole print.

I want to make one last try and ask for your help. Ask me what to provide - I will provide. I can not take it anymore.

The printer is delta.

foosel, 2020-09-24:
For questions or other kind of help or support, please get in touch on the [community forum](https://community.octoprint.org). The bug tracker is reserved for bugs and feature requests, and while there *might* be a big here as the root cause, the bigger likelihood is some issue rooted in your full setup and environment. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md). Thank you! :)

closed 2020-09-24 by foosel

whoim2, 2020-10-04:
its a bug)
old old bug.
See: https://prnt.sc/uszhwc
at this fill 100% freezes, always

cp2004, 2020-10-04:
@whoim2 - if it's a bug, file it as a bug. If not, find support on the community forum.

Read this section 'How to file a bug report' in the [Contributing Guidelines](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)

Then you need to fill in the template provided when you open a bug report, which is also in the contribution guidelines. The only way issues are able to be analysed is if all of the information in the template is filled out. Particularly safe mode.... (*hint at something you might try*)

whoim2, 2020-10-04:
> @whoim2 - if it's a bug, file it as a bug. If not, find support on the community forum.
>
> Read this section 'How to file a bug report' in the [Contributing [...]

Thanks, im create topic on forum