OctoPrint/OctoPrint #3738: [Request] Select the first tool by default, when there is only one tool on the printer

LRTNZ, 2020-09-20:
## Issue/Fustration
I am often fustrated when I go to the control tab of OctoPrint, to run some plastic through my hotend to prime it (Say when changing colour) , by the fact I have to select the tool before I can use it - especially when it is the only tool on the printer.

## Ideal Solution
When there is only one tool on the printer, for that to always be the default selected tool. 

Potentially to build on this, if there are multiple tools, have an easy and obvious way for the user to quickly select and change the default tool (Say you run different nozzle sizes, and are planning to work with one particular size for a few days, having a way inside the control tab to mark the default tool would be great), which will work in the same way, being the one selected by default.

## Altenatives
I am not sure what other alternatives there really are, to a default selection of the one and only tool?

## Additional Context
It is really only an ease of life thing, that I think may remove a tiny bit of fustration from the days of all users that use the control tab to run filament through their extruder.

cp2004, 2020-09-20:
If you *don't* select a tool, I think it extrudes through the previously selected one, which for single extruder is always just T0, so it extrudes anyway. Is this the case for you?
If this is correct, then maybe it is a bug that it shows select tool with just one tool 🤷 
I think the UI with selection could be improved too, since it does not show which tool has been selected:
[image]

jneilliii, 2020-09-20:
I think this would definitely be better to automatically select the first one by default on page load and making the selected one actually show up.

LRTNZ, 2020-09-20:
> If you _don't_ select a tool, I think it extrudes through the previously selected one, which for single extruder is always just T0, so it extrudes anyway. Is this the case for you?

I would test that right now... But I am in the middle of building up my Benchy Armada, so I cannot really do that rn 😅 
> If this is correct, then maybe it is a bug that it shows select tool with just one tool 🤷

Yes, I think that may be the case.
> I think the UI with selection could be improved too, since it does not show which tool has been selected:
> [image]

That's odd... I could have sworn that I have seen the text in the button change to be the tool that I have selected? 
I am using Version 1.4.2, based on what is on the "about OctoPrint" window.

LRTNZ, 2020-09-20:
> That's odd... I could have sworn that I have seen the text in the button change to be the tool that I have selected?

NVM, it does not show the tool that has been selected... Print just detached itself from the bed, so got to test earlier than planned 😢

foosel, 2020-09-21:
Tool selection isn't required, and displaying the currently selected tool there isn't possible as I cannot query that data from the firmware and it can be changed either through the controller or GCODE on the printer's SD and thus without OctoPrint noticing it.

I will see that it is clearer that you only need to select something if you want to change the active tool, and I will also make the whole selection box hide if there's only one.

foosel, 2020-09-21:
Single extruder setup: 

[image]

Multi extruder setup: 

[image]

Popover:

[image]

Will be part of 1.5.0.

LRTNZ, 2020-09-21:
Looks great! I wonder if down the line support for reading the selected tool is something that could be added by the Marlin team? That should cover the majority of printers?

foosel, 2020-09-21:
Possibly yes, alas, the majority of printers out there actually (according to my tracking) do not run mainline Marlin but forks, and from my experience it takes eons for fixes and improvements to make their way first downstream to the forks and then on actual printers. Most users rarely flash, and if so flash whatever their printer's vendor provides them with, which rarely is mainline Marlin.

LRTNZ, 2020-09-21:
> Possibly yes, alas, the majority of printers out there actually (according to my tracking) do not run mainline Marlin but forks, and from my experience it takes eons for fixes and improvements to [...]

Yeah, I can certainly see the issue there - maybe if it got into the Prusa fork, it could get out to at least some printers - as their ones squawk like anything when there is an update available for them 😅

closed 2020-11-30 by foosel

cp2004, 2020-12-01:
According to the linked ticket, RepRapFirmware requires a `T0` command even with just one extruder. This should *either* be done automatically, or an option to re-show the button.

commit dd78fd3e (2020-09-21): 🚸 Clarify tool selection purpose & limitations

Closes #3738