OctoPrint/OctoPrint #3739: [Bug] Webcam snapshots continually fail

KenLucke, 2020-09-21:
#### What were you doing?
Trying to get the webcam/timelapse working as it used to.

1. Set webcam settings as usual, and same as ones that work perfectly in other instances.
2. Wait a little
3. Pretty pink Error Message shows up telling me webcam snapshots are failing
[image]

4. Double check settings
5. Determine settings are identical, except for IP address, to other instances that are working.  (BTW, https:// addresses do not work, no doubt due to authentication failures.  Perhaps add authentication information if https is being used?)

6. Check test image
[image]

7. Lather, rinse, repeat.
8, Give up, file bug report.

#### What did you expect to happen?
That snapshots and timelapses work, as they did before I reinstalled OctoPrint from scratch after downgrading from Python 3 due to too many off my important plugins not working yet.

#### What happened instead?
Failure to create snapshots for timelapse

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Version of OctoPrint
1.4.2

#### Operating System running OctoPrint
Mac OSX 10.15.6

#### Printer model & used firmware incl. version
N/A

#### Browser and version of browser, operating system running browser
N/A

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

KenLucke, 2020-09-22:
Here's a snippet of a relevant section of the log.  There are hundreds of these throughout the log.

```2020-09-21 15:19:24,198 - octoprint.timelapse - ERROR - Could not capture image /home/pi/.octoprint/timelapse/tmp/PLA_Temp_Tower_220deg-225deg_(Bowden)_20200921145700-0.jpg from http://
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/timelapse.py", line 644, in _perform_capture
    verify=self._snapshot_validate_ssl)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 516, in request
    prep = self.prepare_request(req)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 459, in prepare_request
    hooks=merge_hooks(request.hooks, self.hooks),
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/models.py", line 314, in prepare
    self.prepare_url(url, params)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/models.py", line 391, in prepare_url
    raise InvalidURL("Invalid URL %r: No host supplied" % url)
InvalidURL: Invalid URL u'http://': No host supplied
2020-09-21 15:19:24,628 - octoprint.plugins.action_command_notification - INFO - Got a notification: Min:
2020-09-21 15:19:24,631 - octoprint.plugins.actioncommands - INFO - Command received: 'action:notification  Min:'
2020-09-21 15:19:24,632 - octoprint.plugins.actioncommands - ERROR - No command found for 'action:notification  Min:'
2020-09-21 15:19:26,752 - octoprint.plugins.thespaghettidetective - ERROR - Server websocket is closing
2020-09-21 15:19:27,083 - octoprint.plugins.dashboard - INFO - cmd_results: 9V
2020-09-21 15:19:27,395 - octoprint.timelapse - ERROR - Could not capture image /home/pi/.octoprint/timelapse/tmp/PLA_Temp_Tower_220deg-225deg_(Bowden)_20200921145700-0.jpg from http://
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/timelapse.py", line 644, in _perform_capture
    verify=self._snapshot_validate_ssl)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 516, in request
    prep = self.prepare_request(req)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 459, in prepare_request
    hooks=merge_hooks(request.hooks, self.hooks),
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/models.py", line 314, in prepare
    self.prepare_url(url, params)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/models.py", line 391, in prepare_url
    raise InvalidURL("Invalid URL %r: No host supplied" % url)
InvalidURL: Invalid URL u'http://': No host supplied
2020-09-21 15:19:30,159 - octoprint.plugins.dashboard - INFO - cmd_results: 9V
2020-09-21 15:19:32,596 - octoprint.timelapse - ERROR - Could not capture image /home/pi/.octoprint/timelapse/tmp/PLA_Temp_Tower_220deg-225deg_(Bowden)_20200921145700-0.jpg from http://
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/timelapse.py", line 644, in _perform_capture
    verify=self._snapshot_validate_ssl)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 516, in request
    prep = self.prepare_request(req)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/sessions.py", line 459, in prepare_request
    hooks=merge_hooks(request.hooks, self.hooks),
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/models.py", line 314, in prepare
    self.prepare_url(url, params)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/models.py", line 391, in prepare_url
    raise InvalidURL("Invalid URL %r: No host supplied" % url)
InvalidURL: Invalid URL u'http://': No host supplied
2020-09-21 15:19:32,975 - octoprint.plugins.action_command_notification - INFO - Got a notification: Min:
2020-09-21 15:19:32,976 - octoprint.plugins.actioncommands - INFO - Command received: 'action:notification  Min:'
2020-09-21 15:19:32,978 - octoprint.plugins.actioncommands - ERROR - No command found for 'action:notification  Min:'
2020-09-21 15:19:33,221 - octoprint.plugins.dashboard - INFO - cmd_results: 9V
2020-09-21 15:19:36,271 - octoprint.plugins.dashboard - INFO - cmd_results: 9V
2020-09-21 15:19:37,218 - octoprint.timelapse - ERROR - Could not capture image /home/pi/.octoprint/timelapse/tmp/PLA_Temp_Tower_220deg-225deg_(Bowden)_20200921145700-0.jpg from http://
Traceback (most recent call last):
[... cut ...]
    hooks=merge_hooks(request.hooks, self.hooks),
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/models.py", line 314, in prepare
    self.prepare_url(url, params)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests/models.py", line 391, in prepare_url
    raise InvalidURL("Invalid URL %r: No host supplied" % url)
InvalidURL: Invalid URL u'http://': No host supplied
2020-09-21 15:19:41,678 - octoprint.plugins.action_command_notification - INFO - Got a notification: X%s Y%s F3000
```

foosel, 2020-09-23:
That `u` prefix stems from the error message, which shows the representation of the URL, which happens to be a unicode string, hence `u'...'`. So that's not the issue here.

What is weird is that snapshot test works whereas snapshot doesn't. Can you please check your `.octoprint/config.yaml` and share what it says under `webcam.snapshot`? Do not under any circumstances share the whole `config.yaml` file publicly here, it contains secrets.

E.g.

``` yaml
webcam:
    ffmpeg: /usr/bin/avconv
    snapshot: http://127.0.0.1:8080/?action=snapshot # <-- there
    stream: /webcam/?action=stream
```

I just tested this on a local install myself and currently cannot reproduce anything out of the ordinary.

foosel, 2020-09-23:
Also please share a **full** `octoprint.log`.

KenLucke, 2020-09-23:
> That `u` prefix stems from the error message, which shows the representation of the URL, which happens to be a unicode string, hence `u'...'`. So that's not the issue here.

Yeah, kinda figured that 'u' was an artifact from somewhere.

```
pi@octopiYelllow:~/.octoprint $ cat config.yaml | grep -2 "snapshot"
            name: Default
            rotate90: false
            snapshot: http://127.0.0.1:8080/?action=snapshot
            streamRatio: '4:3'
--
webcam:
    ffmpeg: /usr/bin/ffmpeg
    snapshot: http://192.168.1.50:8080/?action=snapshot
    stream: /webcam/?action=stream
    streamRatio: '4:3'
```

I am afraid this (or a full log) is not going to help much, because everything has mysteriously gone back to working properly.  The ONLY thing I did was flash a new firmware to change my right side probe points during a BLTouch probe. (BTW, before the thought that it was a reload/reboot of OctoPrint after a firmware update that did it, I had already previously tried restarting OP, and rebooting the Pi, with no effect)  So, I tried flashing back to the previous version, and it's still working.

I guess the alien spacecraft is no longer hovering over my house.

KenLucke, 2020-09-23:
Here is the full log from when this WAS happening.

https://www.dropbox.com/s/3pmc0kw2m3c9wej/octoprint.log

foosel, 2020-09-24:
Well, since it's working again, I'm closing this. Should the aliens return we can always reopen and I might even get a wig and cosplay as Scully while investigating.

closed 2020-09-24 by foosel