OctoPrint/OctoPrint #3740: "Manage language packs" dialogs appears out of nowhere

frenck, 2020-09-22:
#### What were you doing?

1. Go to OctoPrint Settings
2. Click on Appearance
3. Click on the Title field (place cursor in it)
4. Hit enter.

#### What did you expect to happen?

To save and close the dialog, I guess?
Not entirely sure, but what happened instead, I did not expect.

#### What happened instead?

The "Manage Language Packs..." dialogs opens

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

1.5.0.dev199+g786e2f860

#### Operating System running OctoPrint

Debian Linux 10, up to date with stable apt repos.
Python 2.7.18, installed via Pyenv, running in a virtualenv.

#### Printer model & used firmware incl. version

None, virtual printer, not relevant for bug report.

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

Not relevant for bug report, no information in it.

#### Link to contents of terminal tab or serial.log

Not relevant for bug report, no information in it.

#### Link to contents of Javascript console in the browser

No relevant entries to be found in there.

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

☝️ That line is from the issue template. I'm unsure what I need to do with that? Should there be a checkbox? Or is it a question? Should I answer with "Yes"?

foosel, 2020-09-22:
> ☝️ That line is from the issue template. I'm unsure what I need to do with that? Should there be a checkbox? Or is it a question? Should I answer with "Yes"? Or is it the answer by itself already? :P

The latter 😉 And thanks, that looks indeed like a fun bug. I'm going to trust you with the statement on the logs because you are who you are 😜, plus I'm pretty sure it's a form definition issue strictly on the front-end.

frenck, 2020-09-22:
> I'm going to trust you with the statement on the logs because you are who you are 😜

Well, there aren't any. It is probably just a wrong binding. So, the code fromi user interface perspective, doesn't have an error.

cp2004, 2020-09-22:
Can reproduce, something to do with the enter key binding I suspect...

foosel, 2020-09-23:
Default form submit behaviour was interfering, that should now be solved on `maintenance`. Will be part of 1.5.0.

Guilouz, 2020-11-13:
Just try on latest 1.5.0rc1, this issue is still there

commit c0495d22 (2020-09-23): 🐛 Remove default submit from settings forms

We only want explicitly defined form behaviour.
Otherwise stuff might be triggered by hitting
the enter key that should not be triggered.

Closes #3740
closed 2020-11-30 by foosel