OctoPrint/OctoPrint #3741: Invalid Semver version

frenck, 2020-09-22:
#### What were you doing?

Reading the contribution guidelines, specifically the part on how OctoPrint is versioned.

https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-octoprint-is-versioned

#### What did you expect to happen?

To see a valid Semantic version in the UI.

#### What happened instead?

Seen version number appearing in the UI, that is not according to semver specifications.

It shows: `OctoPrint 1.5.0.dev199+g786e2f860.dirty`

However, the dev marker is preceded with a `.`, which is not valid.

#### Did the same happen when running OctoPrint in safe mode?

uh, yeah?

#### Version of OctoPrint

1.5.0.dev199+g786e2f860

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

Not relevant to this bug report.

#### Browser and version of browser, operating system running browser

Not relevant to this bug report.

#### Link to octoprint.log

Not relevant to this bug report.

#### Link to contents of terminal tab or serial.log

Not relevant to this bug report.

#### Link to contents of Javascript console in the browser

Not relevant to this bug report.

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

☝️ Thanks! for reading it for me 😉

frenck, 2020-09-22:
Interesting, classified as a doc bug. Does that mean the project doesn't follow semver?

foosel, 2020-09-22:
Doc bug then, should say pep440 compliant version with major.minor.patch following semantic versioning rules.

As far as I understand pep440 it would be illegal to do something like 1.5.0dev199 and it actually must be 1.5.0.dev199 to be pep440 compliant, see https://www.python.org/dev/peps/pep-0440/#developmental-releases

frenck, 2020-09-22:
Aah right, yeah pep440 and Semver are not compatible.

foosel, 2020-09-22:
Actually just realized it even says that it follows the semver *scheme* of major.minor.patch, not the full semver spec, and is pep440 compliant. So not a doc bug.

Since it puzzled you: suggestions on how to make this clearer in the wording? Just saying pep440 doesn't imply the contract at the heart of semantic versioning either. And it's actually adhering to that. The whole point of this section is to explain which part of the version number indicates what and which parser should be used for comparisons.

foosel, 2020-09-23:
So, two things. I've moved both the versioning and branching docs into the development docs. I've also changed the wording of the versioning description to hopefully make it clearer that the version *format* is PEP440, while the version *contract* is semver. I gotta admit, I was kinda surprised to see that semver has involved into a full format specification, when I started versioning stuff following the semver *contract* I can't remember that being the case yet.

In any case, see https://docs.octoprint.org/en/master/development/versioning.html and also https://docs.octoprint.org/en/master/development/branches.html.

Additionally I've added a link from the dev docs to the contribution guidelines since you mentioned on Discord you weren't even aware of their existence at first due to jumping in with the dev docs.

Going to close this in the hopes it's cleared up, if not feel free to clarify what is still unclear and I'll look into it.

closed 2020-09-23 by foosel

frenck, 2020-09-23:
Awesome 👍