OctoPrint/OctoPrint #3743: Layer flyout display on scroll bar in GCode viewer in wrong location

chrissv, 2020-09-22:
#### What were you doing?

1. Open up a new tab in Chrome, go to Octoprint
2. "Load" a GCode file (without printing)
3. Go to the GCode viewer. Wait for the model to load
4. Hover mouse over the Z scrollbar handle.

#### What did you expect to happen?
I expect to see the current flyout display for the Z position located at the Z scrollbar handle.

#### What happened instead?
The current flyout display for the Z position is located at the very top of the Z scrollbar area.
Note: if I move the Z position handle, the flyout display correctly tracks the handle position thereafter.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.4.2, but it has happened in all versions of Octoprint for the past couple of years.

#### Operating System running OctoPrint
Raspberry Pi Jessie (latest updates applied)

#### Printer model & used firmware incl. version
Prusa I3 clone, Merlin 1.1.8

#### Browser and version of browser, operating system running browser
Windows 10 running Chrome Version 85.0.4183.102 (but it happen in all versions I've tried)

#### Link to octoprint.log
Not collected.

#### Link to contents of terminal tab or serial.log
No content - model wasn't printing

#### Link to contents of Javascript console in the browser
https://pastebin.com/jzqVgiYk

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2020-09-23:
This is now fixed on `maintenance` and will be included in 1.5.0.

chrissv, 2020-09-23:
Awesome!

commit d30b3a39 (2020-09-23): 🐛 Fix tooltip on freshly initialized slider

Closes #3743
closed 2020-11-30 by foosel