OctoPrint/OctoPrint #3749: Missing power/restart button on the navigation bar

theone11, 2020-09-25:
#### What were you doing?

Missing power/restart button on the navigation bar
Clean install showed the button but after installing some plugins I noticed it was gone.
Don't remember exactly after which plugin it disappeared.

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

SafeMode also doesn't show the button.

#### Version of OctoPrint

OctoPrint 1.4.2 Python 3.7.3

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

Ender3v2

#### Browser and version of browser, operating system running browser

Chrome, Edge, IE

#### Link to octoprint.log

https://community.octoprint.org/t/issue-no-power-button-on-the-octoprint-webui/24510

#### Link to contents of terminal tab or serial.log

https://community.octoprint.org/t/issue-no-power-button-on-the-octoprint-webui/24510

#### Link to contents of Javascript console in the browser

https://community.octoprint.org/t/issue-no-power-button-on-the-octoprint-webui/24510

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

foosel, 2020-09-25:
As it was caused by a third party plugin according to your own statement this is not an OctoPrint bug. The plugin probably need your configuration. Re-add the village under settings - server

closed 2020-09-25 by foosel

theone11, 2020-09-25:
But I disabled all plugins and also tried safemode and it didn't help.
What do you mean by "Re-add the village under settings - server"?
I have plugin blacklist processing enabled from the installation wizard, Only using plugins from official repository.

jneilliii, 2020-09-25:
She's referring to these settings.

[image]

foosel, 2020-09-25:
Yeah, and "village" should have been "commands" - stupid auto correct again -_-

Safe mode can't work miracles - if some settings get nuked by a plugin they get nuked. Disabling the plugin won't magically bring them back. And this stuff is configured in the settings.

theone11, 2020-09-25:
Thanks again,
I didn't realize that these entries basically create the power/restart button and if they are empty then it doesn't appear. It makes sense in hindsight but maybe a comment in the settings regarding this could help others in the future.
Something like "If these command are empty then power/restart button will not appear in the WEBUI".