OctoPrint/OctoPrint #3751: Parametric feedrate and emergency stop button

federicodilo, 2020-09-26:
Hello,
I noticed that by clicking on the control buttons will issue movements in XY or Z axis with a fixed feedrate (eg. G0 Z10 F2) possibily resulting in slow movement speed.
Maybe, as did in Pronterface, an input box would be the most comfortable solution in order to modify feedrate dinamically without the help of any multiplier:
[image]
Nevertheless homing feedrate should remain the same.

Also an emergency stop button in control tab (force disconnection) would be useful in order to prevent hazardous events while moving.

cp2004, 2020-10-23:
Just looking at this request, and wanted to clarify a couple of things:
* There is already emergency stop plugins available (either web UI or via GPIO) - https://plugins.octoprint.org/search/?q=emergency+stop - they might help for that part of the request.

* WRT to the feedrate - it uses the number set by you in the printer profile, to determine the feedrate there, so it is not 'fixed' and can be changed. What is *not* available is setting the feedrate in the control tab *per move*. If that is what this request is about, then let me know.

foosel, 2020-11-02:
@federicodilo a response from you is required here.

federicodilo, 2020-11-03:
Hello, sorry, I've been quite busy these days.
Thanks for the response, I stupidly forgot about the form of the printer profile wondering why my printer was moving so slowly.
Feel free to close the issue if you prefer.
Sorry and thank you.

cp2004, 2020-11-03:
Thanks for getting back to us, since this request is no longer applicable I'll close it.

closed 2020-11-03 by cp2004