OctoPrint/OctoPrint #3754: Request/safety - Enforce unique watched and upload paths.

paukstelis, 2020-09-30:
#### What were you doing?
Witnessing people play around with uploads and watched paths. If upload and watched paths are the same, bad things happen.

#### What did you expect to happen?
Path changes.

#### What happened instead?
Bad things. Having the watched and uploads path the same wipes out all files. Perhaps these two paths should be forced to be unique from one another? Inexperienced users might be messing around with this, not even knowing what 'watched' is for.

#### Did the same happen when running OctoPrint in safe mode?
Didn't try.

#### Version of OctoPrint
All versions.

#### Operating System running OctoPrint
All OS

I have read the FAQ.

foosel, 2020-10-09:
Starting with OctoPrint 1.5.0, OctoPrint will 

  * detect duplicate folder paths in the settings dialog, mark them up and refuse to save them
  * detect duplicate folder paths on the settings API and throw an error back instead of saving them
  * detect duplicate folder paths on startup and refuse to start, exiting with a fatal invalid config error

paukstelis, 2020-11-18:
Thank you and happily closing. Well done!

commit b96f7657 (2020-10-09): 🥅 Detect and deny duplicate folder path config

Closes #3754
closed 2020-11-18 by paukstelis