OctoPrint/OctoPrint #3758: Custom Controls Behavior Not Matching Documentation

jneilliii, 2020-10-06:
#### What were you doing?

Adding custom controls to OctoPrint either via config.yaml or getAdditionalControls does not follow current documentation on enabled behavior. 

#### What did you expect to happen?

Adding custom controls without an enabled property defined to disable itself while printer is disconnected (working) and while printing (not working). 

#### What happened instead?

The buttons added via config.yaml or getAdditionalControls like in my BLTouch plugin [here](https://github.com/jneilliii/OctoPrint-BLTouch/blob/dcd8475aa9c6d8af3d88d0b1c55fa78ad0092eb9/octoprint_BLTouch/static/js/BLTouch.js#L22) remain enabled while printing.

#### Did the same happen when running OctoPrint in safe mode?

N/A

#### Version of OctoPrint

1.4.2

#### Operating System running OctoPrint

OctoPi 0.18 nightly build, python 3.7.3

#### Printer model & used firmware incl. version

Virtual Printer

#### Browser and version of browser, operating system running browser

Google Chrome Version 85.0.4183.121 (Official Build) (64-bit)

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

No errors.

#### Screenshot(s)/video(s) showing the problem:

[image]

Notice Get Position button (from config.yaml) and BLTouch buttons (from getAdditionalControls) are all enabled while printing.

I have read the FAQ.

foosel, 2020-10-08:
Documentation bug as well. It was never implemented the way it's stated in the docs there, so I've fixed them. The idea is that only some custom controls require disabling during printing (e.g. you don't want to allow the user to send arbitrary movement related commands, but fan control is fair game). The ``enabled`` prop is in place to modify default behaviour.

closed 2020-10-08 by foosel

jneilliii, 2020-10-08:
So the recommendation here would also be a settings overlay instead of getAdditionalControls to have a dynamically enabled button?

foosel, 2020-10-08:
No, getAdditionalControls should work here too unless you specifically require feedback functionality. But the `enabled` property needs to be set accordingly, either to a string with the bindings (the view model is available via the context) or to a function taking in the bindings. 

[image]

Also: 

https://github.com/OctoPrint/OctoPrint/blob/90130dcd7d490bb6bf357d2d269556856e202bdd/src/octoprint/static/js/app/viewmodels/control.js#L222-L231

commit d62b9f00 (2020-10-08): 📝 Fix documented default enable logic on custom controls

They are only disabled when the printer is disconnected, not when it
is printing. This behaviour can be overridden with the `enabled`
property.

Closes #3758