OctoPrint/OctoPrint #3759: getAdditionalControls regex and template don't work

jneilliii, 2020-10-06:
#### What were you doing?

Trying to assist a user on the community forum [here](https://community.octoprint.org/t/trying-to-get-positional-data-with-m114/24832?u=jneilliii) with getAdditionalControls and regex template matching, came to the conclusion that it does not work as expected. The regex is not seen/matched and output is not displayed on the UI when the button is pressed.

Example config.yaml.
```
controls:
-   children:
    -   command: M114
        name: Get Position
        regex: X:([-+]?[0-9.]+) Y:([-+]?[0-9.]+) Z:([-+]?[0-9.]+) E:([-+]?[0-9.]+)
        template: 'Position: X={0}, Y={1}, Z={2}, E={3}'
    name: Reporting
```

equivalent getAdditionalControls callback.

```
self.getAdditionalControls = function () {
    return [{
            name: 'Reporting',
            children: [{
                    command: 'M114',
                    name: 'Get Position',
                    regex: 'X:([-+]?[0-9.]+) Y:([-+]?[0-9.]+) Z:([-+]?[0-9.]+) E:([-+]?[0-9.]+)',
                    template: 'Position: X={0}, Y={1}, Z={2}, E={3}',
                }
            ]
        }
    ];
}
```

#### What did you expect to happen?

To work exactly the same way that Custom Controls work from config.yaml.

#### What happened instead?

Command is sent, but template is not filled out and displayed.

#### Did the same happen when running OctoPrint in safe mode?

Can't reproduce without a plugin.

#### Version of OctoPrint

1.4.2

#### Operating System running OctoPrint

OctoPi 0.18, Python 3.7.3

#### Printer model & used firmware incl. version

Virtual Printer

#### Browser and version of browser, operating system running browser

Google Chrome Version 85.0.4183.121 (Official Build) (64-bit)

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

No errors.

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

jneilliii, 2020-10-06:
Does the regex have to be defined in the callback as an actual regex rather than a string representation of the regex?

foosel, 2020-10-08:
I fear that's more a doc bug than anything. The way the feedback controls work is that the backend has to know about them so it can scan for them. Injecting additional ones only on JS side doesn't register them where they need to get registered in that case, since the backend doesn't even know about their existence, and thus the regex will never be matched against and no results will ever bubble up to the frontend. 

I'll add a note on that to the docs. The solution for a plugin here would be a settings overlay I think.

commit 1688d8f2 (2020-10-08): 📝 Feedback controls only work if defined in config.yaml

Closes #3759
closed 2020-10-08 by foosel