OctoPrint/OctoPrint #3763: Easier Batch Plugin Installation From Plugin Manager [Request]

CAP1Sup, 2020-10-08:
**Is your feature request related to a problem? Please describe.**
Whenever configuring Octoprint, installing all the plugins is a pain due to clicking install, then waiting for each to install individually.

**Describe the solution you'd like**
It would be nice to have check boxes instead. Then, once you had decided on which you wanted to install, you could click the install all button and then they would all batch install.

**Describe alternatives you've considered**
N/A

**Additional context**
It'd just be really nice to have. The install process takes over an hour or two with the plugins I like.

jneilliii, 2020-10-08:
I have to assume that length of time probably includes the Bed Level Visualizer plugin and the numpy build dependency taking a long time. This gets much better in Python 3 environment that OctoPrint is currently supporting and headed toward. I also recommend using backup/restore for doing a lot of plugin installs at once.

cp2004, 2020-10-08:
Kind of similar to discussion in #3711 about extending the backup plugin to create the `plugins.json` file on its own, which could then be uploaded to bulk install. To then install multiple, you could either add them to this file, or (to not try editing files, since that is not great UX) have checkboxes like you say.

LazeMSS, 2021-03-31:
Is this not a duplicate of #2626