OctoPrint/OctoPrint #3765: [maintenance] [bug] Plugin Release Channels Not Sticking

jneilliii, 2020-10-08:
#### What were you doing?

Enabling release channels for my plugin for OctoPrint 1.5.0 and when I switch one of the plugin's release channels to another channel the ones already set revert to Stable.

#### What did you expect to happen?

Release channel setting to stay untouched.

#### What happened instead?

Release channels getting changed while changing others.

#### Did the same happen when running OctoPrint in safe mode?

N/A

#### Version of OctoPrint

OctoPrint 1.5.0.dev248+g60f45ffaf 

#### Operating System running OctoPrint

Windows dev environment running Python 3.7.2

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Google Chrome Version 85.0.4183.121 (Official Build) (64-bit)

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

No errors.

#### Screenshot(s)/video(s) showing the problem:

[image]

video doesn't show the menus, but I promise they are there.

I have read the FAQ.

foosel, 2020-10-12:
Pretty sure that this one and #3767 have the same underlying cause, as both those features are implemented in the same endpoint in which I've observed some issues with config caching in `self._settings` that I'll need to investigate further.

foosel, 2020-10-14:
Should be fixed on `maintenance`.

commit e7d037d3 (2020-10-14): 🐛 SWU: Fix issue with check config persistence from plugins

If we want to be able to in-place overlay the effective config, we MUST
do a deepcopy of the default or we'll actually operate on the default,
effectively leading to any changes to plugin check configs to be cleaned
up from the config file.

Closes #3765
Closes #3767
closed 2020-10-14 by foosel