OctoPrint/OctoPrint #3766: [Request] see who cancelled the print

mrseeker, 2020-10-09:
**Is your feature request related to a problem? Please describe.**
It is not clear which plugin causes a failure of a print. On Discord I usually see requests that range something from "my print got cancelled midprint" to "what causes octoprint to cancel prints"

**Describe the solution you'd like**
When a print is cancelled, a reason in the logging why it's cancelled (eg: "octoprint.ui.cancelbutton", "plugins.filamentsensor", "printer.error.disconnected")

**Describe alternatives you've considered**
Closing 20 plugins to find the source of the issue

**Additional context**
Add any other context or screenshots about the feature request here.

foosel, 2020-10-19:
Makes sense, added for 1.5.0.

```
2020-10-19 11:01:23,976 - octoprint.util.comm - INFO - Cancelling job on behalf of plugin virtual_printer
[...]
2020-10-19 11:04:38,686 - octoprint.util.comm - INFO - Cancelling job on behalf of user admin
```

foosel, 2020-11-30:
1.5.0 has just been released.

closed 2020-11-30 by foosel