OctoPrint/OctoPrint #3767: [Maintenance/1.5.0-dev] Only the first ignored SW Update notification is saved

cp2004, 2020-10-09:
#### What were you doing?

1. Head to the 'Software Update' settings in the UI
2. Disable update check for a plugin
3. Disable check for *another* plugin
4. Restart server
5. View problem: the update notification comes back *only for the 2nd plugin*

After changing the first value, the second does not seem to be written to disk. For example, I played with 3 plugins that need updates:
* Marlin EEPROM Editor (1.2.1 --> 2.0.0) - note that this is development install, would not work in production
* WS281x LED Status (0.5.0 --> 0.5.1) - also development install
* Ngrok (0.2.1 --> 0.3.0) - production install (plugin manager)

Observing the config.yaml, only the first gets an entry under plugins.softwareupdate.checks:
```yaml
checks:
    eeprom_marlin:
        disabled: false
```

EEPROM Marlin is able to be changed between true/false, but the others never appear. They do persist across reloads of the Web UI, so I think it is just the writing to disk

#### What did you expect to happen?

Settings would stay as they have been set

#### What happened instead?

Reverted to default (enabled)

#### Did the same happen when running OctoPrint in safe mode?

Unable to test, since safe mode means plugin version checks are disabled.

#### Version of OctoPrint
`1.5.0.dev271+gf291fa898`

#### Operating System running OctoPrint

OctoPi 0.17, but with development environment installed.

#### Printer model & used firmware incl. version
N/A

#### Browser and version of browser, operating system running browser
New Edge, 85.85.0.564.70

#### Link to octoprint.log

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5356409/octoprint.8.log)
^ this is a bit long, but no error was produced when the API endpoint was called - it seemed to think everything was successful. At least as far as I could see.

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

[browser-log.log](https://github.com/OctoPrint/OctoPrint/files/5356447/browser-log.log)

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2020-10-10:
I have a feeling that this might have been an issue introduced in a subsequent commit, since it doesn't work on my windows machine at all, and the report above was from my dev RPi, where I used the feature when it was first a thing on maintenance.

cp2004, 2020-10-11:
Ok so I have played with this on and off, and some steps to reproduce *seem* to be either:
1. Mute notifications for 1 plugin, save
2. Restart server
3. Observe notification again
OR:
1. Mute notifications for 2 plugins, save
2. Restart server
3. Observe 1 was saved, 2nd not
4. (optional) Mute 2 more plugins, notice that it will now only allow one, and switch between the two:

[image]

foosel, 2020-10-12:
Pretty sure that this one and #3765 have the same underlying cause, as both those features are implemented in the same endpoint in which I've observed some issues with config caching in `self._settings` that I'll need to investigate further.

foosel, 2020-10-14:
Should be fixed on `maintenance`.

commit e7d037d3 (2020-10-14): 🐛 SWU: Fix issue with check config persistence from plugins

If we want to be able to in-place overlay the effective config, we MUST
do a deepcopy of the default or we'll actually operate on the default,
effectively leading to any changes to plugin check configs to be cleaned
up from the config file.

Closes #3765
Closes #3767
closed 2020-10-14 by foosel