OctoPrint/OctoPrint #3768: [Request] Move "Cear Notifications" Button to top

KenLucke, 2020-10-09:
**Is your feature request related to a problem? Please describe.**
LOL, Self Explanatory:  :)
https://www.dropbox.com/s/c9rd1cusmq3aefd/screencast%202020-10-09%2015-42-12.mp4

**Describe the solution you'd like**
Move the Clear Notifications button to the top of the list instead of the bottom.
Maybe add optional regexes to remove frequent [and unneeded] notifications from being added, and/or a max number of notifications parameter

**Describe alternatives you've considered**
Not using it

**Additional context**
Add any other context or screenshots about the feature request here.

jneilliii, 2020-10-10:
LOL, that's a funny one Ken. Obviously you're getting flooded by M117 commands with action prompt support enabled on your firmware. I like your suggestion of putting a clear in the sidebar header like the file list, and filtering like the terminal output would make sense with optional regexs and max limits.

cp2004, 2020-10-10:
I have fixed the scrolling area, so it looks the same as the filemanager does for the next release. So this shouldn't be much of an issue in future, as you can scroll past them easily. (since its limited to 300ish px iirc)

#3718

KenLucke, 2020-10-11:
> I have fixed the scrolling area, so it looks the same as the filemanager does for the next release. So this shouldn't be much of an issue in future, as you can scroll past them easily. (since its limited to 300ish px iirc)

Thank you.  That will definitely help.   It would still be nice some time in the future to see some sort of limit or filter.

foosel, 2020-10-14:
[image]

Will be in 1.5.0, as will a sort order toggle as visible above.

foosel, 2020-11-30:
1.5.0 has just been released.

closed 2020-11-30 by foosel