OctoPrint/OctoPrint #3773: [Request]Potential workaround for Marlin inabilty to cancel M190/M192 without an M112

psonoda, 2020-10-14:
Problem to solve: Having to wait for hotend or bed heat up after cancelling a print job.
Alternative: Waiting or using Emergency Stop (M112) and reconnecting.

I realize that this is not an Octoprint issue and a true fix lies within Marlin, but I suspect the sweet release of death will take me before Marlin changes this behavior...

As a setable feature within Octoprint automatically capture all M190/M192 commands and modify them to M140/M104 and then have Octoprint pause the gcode processing until the appropriate temperature is reached. 
One twist to the above is you could wait until temperature was at let say 90% of the desired temperature and then issue the M190/M192. That way you could get the AutoTemp options to take affect.

I know this is a less than perfect solution but it would make a cancel print much more responsive. I briefly looked at making a plugin for this but I didn't see any ability to pause the gcode output and wait for a certain temperature but I may have missed it.

jneilliii, 2020-10-14:
Isn't that what this plugin does?

https://plugins.octoprint.org/plugins/setandwait/

psonoda, 2020-10-14:
You're absolutely right! Thank you!

closed 2020-10-14 by psonoda

foosel, 2020-10-14:
You can also configure marlin to support M108. If it reports that support to OctoPrint, OctoPrint will send it to the printer on a cancel.