OctoPrint/OctoPrint #3774: Printing with non-default filament

carstenpoulsen, 2020-10-14:
#### Trying to print with single filament other than #1

I use PrusaSlicer to prepare gcodes for Prusa MK3 w/ MMU
If I have a single color/filament model and choose extruder #1 the model is printed equally well on SD and by using OctoPrint on a Raspberry PI 3.
If I slice the same model and choose extruder other than #1 (e.g. #3) the model is printed well on SD but when using OctoPrint the filament is never loaded but the printer continues as if the filament is loaded.

When printing multicolor models where #1 is the first extruder to be used, there is no problem loading the extruder and change to any different extruder during print.

#### What did you expect to happen?

I expect the same behavior when using OctoPrint as when using SD card. i.e. that the filemant is laoaded

#### What happened instead?

The filament is never loaded, but the printer continues as if it is loaded.

#### Did the same happen when running OctoPrint in safe mode?

The same behavior as in non-safe mode.

#### Version of OctoPrint

OctoPrint 1.4.0 running on OctoPi 0.17.0

#### Printer model & used firmware incl. version

Prusa3d/MK3
Firmware: 3.8.1-2869
MMU connected
FW: 1.0.6-372

#### Browser and version of browser, operating system running browser

Chrome

#### Link to octoprint.log

http://boeghpoulsen.dk/octoprint.log
http://boeghpoulsen.dk/octoprint.log.2020-10-08

#### Link to contents of terminal tab or serial.log

http://boeghpoulsen.dk/serial.log
http://boeghpoulsen.dk/serial.log.2020-09-17_10-05-47

#### Link to contents of Javascript console in the browser

n/a problem seen in different browsers

#### Screenshot(s)/video(s) showing the problem:

Video not needed:
When using extruder #1, there is no issue.
When using extruder #3 (or other extruder than #1) the filament is not loaded.
When using SD card there is no issue disregarding what extruder.

I have read the FAQ.

foosel, 2020-10-14:
Please take a look at https://plugins.octoprint.org/plugins/mmu2filamentselect/ and see if it helps. If it doesn't, *enable* `serial.log`, reproduce the issue and share the resulting log. Your shared once are empty apart from a message telling you to enable the log. Also share a GCODE file with which to reproduce.

Last I heard, the MMU is a finicky beast and requires some manual intervention for single extruder prints via serial. That is not an issue located in OctoPrint but rather the way the printer and printer firmware function. Also make sure your generated GCODE actually contains a tool selection command. OctoPrint can't know that there's an MMU on the other end, it will pipe your GCODE through to the printer and expect it to properly process it. If GCODE works from SD which doesn't work from OctoPrint, that's due to the firmware doing some stuff with files from SD it doesn't do with commands on serial. So you might have to adjust your slicer, or use a plugin.

carstenpoulsen, 2020-10-14:
I think that the filamentselect is useful if you do not have slicer support for filament selection.
In my case I slice and prepare everything in several files. Introducing a filementselector merely incur a point of error.

I have replaced the serial.log.
I forgot the gcode link:
http://boeghpoulsen.dk/CentralUnitSpacer.gcode
I have reviewed the gcode.
As it works perfect on SD it indicates that something is rotten in the serial connection.
I have tried to convince Pruse that here is a deficiency, but they point at octopi.

I think that the prusa software leaves much to desire.
I program embedded systems myself and it is immediately apparent that prusa don't.

foosel, 2020-10-14:
```
2020-10-14 11:49:55,184 - Not queuing T2, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware
```

How have you set up your printer profile in OctoPrint? Make sure you have set it to the number of extruders and a shared nozzle. Otherwise OctoPrint will refuse to send tool selection commands for tools it doesn't know about, because it can cause freezing firmware on various printers out there to send them a tool command for a tool that doesn't exist.

There's a screenshot on how things should look like here: https://community.octoprint.org/t/i-installed-octoprint-1-3-12-or-later-and-now-my-mmu2-doesnt-work-properly-the-whole-printer-resets-how-to-fix-this/12962

carstenpoulsen, 2020-10-14:
OK. this is new.
I did not care about setup, because I assumed this was to enable some automation I did not care about.
I use octoprint as SD replacement.
And the fact that at some point it worked, I had no clue that octopi setup could influence MMU functionality.
Add to this that MMU works fine if just I have #1 as the first filament.

I will test with a proper printer profile.

foosel, 2020-10-14:
> Add to this that MMU works fine if just I have #1 as the first filament.

That makes total sense if you have your printer profile misconfigured, as OctoPrint will not refuse to send a `T0` command with only one configured extruder, however it WILL refuse to send a `T2` command.

The printer profile is not just for cosmetics ;)

carstenpoulsen, 2020-10-14:
When does octoprint refuse to send T2 command?
If I have a gcode file with both T0 and T2, the print is fine ... ok, I need to test that again,
There may be something there. I maight have encountered problems with two-color setups. Evidence missing.

This is equivalent to changing some radius on my design because OctoPrint think that they know better!!

Then another question: It would be decent if OctoPrint issued an error message rather than just editing MY gcode.
Noone should ever edit or omit anything from my files at least unless I am informed.
To be honest, I am annoyed when someone changes my files without my permission or consent.
I mean: I made the design so I have the copyright.

Does Octoprint make other changes?
And why on earth is T2 (or other) removed?
It does not make any sense.

carstenpoulsen, 2020-10-14:
OK. Now it works.
I think I will stop using octoprint as I consider it unreliable software. Remove some command without noticing renders OP unreliable.
If OP another time make changes I will have no clue and spend hours figuring out why the function has changed or ceased.
I am no longer able to support because it has costed me effort and lost revenue.

foosel, 2020-10-14:
Do whatever works for you. This is free software, no one forces you to use it.

closed 2020-10-14 by foosel

carstenpoulsen, 2020-10-14:
Sure not.

But it is sad that bad software management results in useless systems.

foosel, 2020-10-14:
You didn't configure your setup properly. But that's obviously my fault, a reason to call something I've worked on for years unreliable and my work on it bad. Thank you for this wonderful demonstration of user behaviour that makes open source maintainers burn out.