OctoPrint/OctoPrint #3777: Force Check for Plugin Updates does not always detect updates.

KenLucke, 2020-10-16:
#### What were you doing?
On 2 identical printers using identical setups of Octoprint, I often get a notice on one that there is an update to a plugin, but no notice on the other.  Knowing that there is a parameter in the Plugin Manager settings that controls how often it checks (default of 6 hours (360 minutes), I have tried using the "Force Update Check" - and it STILL fails to detect the update, even when setting the TTL to 1 minute (or Zero), and even across OP restarts.

#### What did you expect to happen?
"Force Update Check" should, well, force an update check, overriding any TTL set in the Plugin Manager settings.

#### What happened instead?
It doesn't.

#### Did the same happen when running OctoPrint in safe mode?
No

#### If you can't test this in safe mode, state why.
Plugin Management functions severely disabled in Safe Mode

#### Version of OctoPrint
1.4.2

#### Operating System running OctoPrint
Mac OS X 10.15.7

#### Printer model & used firmware incl. version
N/A

#### Browser and version of browser, operating system running browser
N/A, but Firefox 81.0.2 (64-bit) & Safari 14.0 (15610.1.28.1.9, 15610)

#### Link to octoprint.log
Not all logs I had enabled actually have any data in them, apparently.  
[image]
The ones that do are at:
<script src="https://gist.github.com/KenLucke/3800e3d1cadd5aa5d05125e33ec20014.js"></script>

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

KenLucke, 2020-10-16:
This from one printer:
[image]

This from the other, even setting TTL to 1, waiting more than 1 minute, then clicking "Force Update":

[image]

cp2004, 2020-10-16:
Despite the idea of 'force-checking' there is rate limiting in place on the GitHub API for unauthenticated users, so it's likely that's what you're running into there. Give it a couple of hours, and it might work again. Probably caused by setting the cache times low - 1 min (I think) means it will ping each plugins' repository *every minute* which will very quickly be rate limited.

KenLucke, 2020-10-16:
Might be, but I didn't set the rate that low until it failed to force the update (and, just as a trial, and has since been restored to default), so that test fails.

KenLucke, 2020-10-16:
Additionally, it now DID show the update, exactly the 6 hours later after my original attempt at force-checking.  BTW, OP had been running the whole time, although no print was in process, so that TTL should have expired long before then.

Force check should override the TTL, not honor it.  If I get a notice on one OP iteration, I should be able to go to any other [idle] one, do a force check, and update appropriately.

jneilliii, 2020-10-16:
I do this regularly when developing plugins and have only had an issue when hitting the github API rate limits, internal network caching (ie proxy server), github API outages, or the pi was unable to resolve hosts correctly. Next time it happens check your developer console for any possible errors, specifically on the network tab where you will see the request happening.

foosel, 2020-10-19:
In your screenshot it looks like it couldn't fetch *any* kind of update information from the backend ("Available: -"), so my money would be on hitting a rate limit there too. Either that, or there was a Github API hiccup (happens way more often than I expected).

Please enable DEBUG logging for `octoprint.plugins.softwareupdate` in the Logging settings:

[image]

Then try to reproduce. It will cause log output like this to be produced:

```
2020-10-19 14:01:09,839 - octoprint.plugins.softwareupdate - DEBUG - Looks like we are online
2020-10-19 14:01:09,958 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Github rate limit: 57/60, reset at 2020-10-19 13:01
2020-10-19 14:01:09,959 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Target: firmware_check, local: 2020.9.23, remote: 2020.09.23
2020-10-19 14:01:09,968 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Github rate limit: 55/60, reset at 2020-10-19 13:01
2020-10-19 14:01:09,969 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Target: file_check, local: 2020.8.7, remote: 2020.08.07
2020-10-19 14:01:10,097 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Github rate limit: 58/60, reset at 2020-10-19 13:01
2020-10-19 14:01:10,098 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Target: curalegacy, local: 1.1.1, remote: 1.1.1
2020-10-19 14:01:10,149 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Github rate limit: 56/60, reset at 2020-10-19 13:01
2020-10-19 14:01:10,151 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Target: filamentmanager, local: 0.5.2, remote: 1.6.3
2020-10-19 14:01:10,205 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Github rate limit: 59/60, reset at 2020-10-19 13:01
2020-10-19 14:01:10,209 - octoprint.plugins.softwareupdate.version_checks.github_release - DEBUG - Target: octoprint, local: 1.5.0.dev290+g2acf4764c.dirty, remote: 1.4.2
```

As you can see, Github rate limits to 60 requests per hour for anonymous requests (the "until" there is UTC, local is CEST, so UTC+0200), which is also the reason why the default TTL is chosen the way it is.

Exceeded API limit will make it look like this:

[image]

foosel, 2020-10-19:
Just to avoid any kind of confusion with that in the future, just pushed this for 1.5.0:

[image]

Still would like confirmation though that the underlying issue here is indeed rate limiting.

KenLucke, 2020-10-19:
> Please enable DEBUG logging for `octoprint.plugins.softwareupdate` in the Logging settings:

Will do.  I thought I had enabled all the update logging items, but I guess I missed that one.  The involved printers are in the middle of 4-5 day prints, so I will not be able to investigate further until one or more of them are finished.

I am ass/u/ming that the rate limiting is by my actual outside IP, rather than an API ID or other specific-to-the-iteration identifier?  So that inquiries from any machine here are viewed as from one source on the other end?

The alert/warning you've added will certaintly eliminate any confusion as to the issue, if that is, indeed, what it is - as seems likely.

jneilliii, 2020-10-19:
You are correct, the rate limiting is not associated to an api key, and therefore reverts to identifying by public ip address.

foosel, 2020-10-19:
Yes, I'm assuming that as well. I've also added a way to provide a personal access token so that if set the requests won't be made anonymously anymore with a rate limit of 60r/h but instead 5000r/h. Will require a manual editing of config.yaml though, that is a bit too niche of a user case to expose in the UI.

KenLucke, 2020-10-19:
> Yes, I'm assuming that as well. I've also added a way to provide a personal access token so that if set the requests won't be made anonymously anymore with a rate limit of 60r/h but instead 5000r/h. [...]

Cool.   You Devs/Authors are awesome. :)

Let me say (for all those that don't), THANK YOU for what you do.

KenLucke, 2020-10-19:
> Yes, I'm assuming that as well. I've also added a way to provide a personal access token so that if set the requests won't be made anonymously anymore with a rate limit of 60r/h but instead 5000r/h. [...]

Considering I have 26 plugins installed, plus OP and its sub-parts themselves being checked, and as it appears from your screen shot that it is checking once per plugin/mains, just two of my printers checking once each would total more than the 60/hr limit, so no wonder it's apparently limiting me. :)

cp2004, 2020-10-19:
Based on that information.... I've marked it as a solved improvement. Might still be worth the debug logging just to be sure that this is entirely the issue, shouldn't be too difficult to reproduce (force check a couple times with logging on)

KenLucke, 2020-10-19:
> Will require a manual editing of config.yaml though, that is a bit too niche of a user case to expose in the UI.

When the update is pushed out, will this be mentioned in the release notes, or can you tell me how to do it here?

cp2004, 2020-10-19:
> will this be mentioned in the release notes

can't say on this, but the config would be in `plugins.softwareupdate.credentials.github`, so in config.yaml:
```yaml
plugins:
  softwareupdate:
    credentials:
      github: xxxxxxxxx
```
obviously inside all the rest of the settings there

KenLucke, 2020-10-19:
> > will this be mentioned in the release notes
> 
> can't say on this, but the config would be in `plugins.softwareupdate.credentials.github`, so in config.yaml:
> 
> ```yaml
> plugins:
>   softwareupdate:
>     credentials:
>       github: xxxxxxxxx
> ```
> 
> obviously inside all the rest of the settings there

OK, thanks.  Is there a format to the credentials, or just any random, password-type string?

cp2004, 2020-10-19:
It is the string you get from creating a personal access token, by heading to Profile > Settings > Developer > Personal access tokens

Don't think it needs any extra permissions, since it is just not anonymous.

[image]

KenLucke, 2020-10-19:
Thanks.  Makes more sense than just a random string, because a bot could just do that, too.  Unfortunately,  I'm not a developer :(

I'll leave you alone now :)

KenLucke, 2020-10-19:
Never mind, I found it will allow me to make a personal token even while not being an actual developer.

foosel, 2020-10-20:
> will this be mentioned in the release notes

Just to answer that one - yep.

foosel, 2020-11-30:
1.5.0 has just been released.

closed 2020-11-30 by foosel

KenLucke, 2020-12-02:
> > can't say on this, but the config would be in `plugins.softwareupdate.credentials.github`, so in config.yaml:
> > ```yaml
> > plugins:
> >   softwareupdate:
> >     credentials:
> >       github: xxxxxxxxx
> > ```
> > 
> > 
> > obviously inside all the rest of the settings there

BTW, this works perfectly.  I was getting the "Rate Limit" error, added my personal credentials to config.yaml, and the rate limiting went away.

bjhiltbrand, 2020-12-11:
I've been having issues with rate limits since updating. I went to add my token to config.yaml, but I want to make sure scopes are correct on github. Which scopes should I choose for this? Thanks.

foosel, 2021-01-25:
Just a public access token with no explicit scopes selected should already suffice.