OctoPrint/OctoPrint #3778: Updates call Restart automatically

CmdrCody51, 2020-10-17:
## What were you doing?

Log in - Normal running - not printing - Announcement of Plugin Update

Select Update Now
Complete the Update and system automatically Restarts.

#### What did you expect to happen?

It depends. If there was one update I could see it doing an unconfirmed restart if one was required by the plugin.
If there are two or more updates I would expect OctoPrint to wait until I decided to restart or all updates are done.
I seem to remember that is the way it worked in the past. At least it wouldn't just do the Restart on it's own.

#### What happened instead?

First time I noticed this there were three updates announced. I had to Update - it Restarted, Update - it Restarted and
finally Updated and it Restarted. It seemed odd.
After discussing this on Discord, I took the next Update and used the Update Now button. When I add to this Issue
I will use the Plugin Manager.
I did notice (as I stayed on the page) that two update errors appeared behind the Reload UI Now popup and could not
get a screenshot fast enough. Sorry.

#### Did the same happen when running OctoPrint in safe mode?

I don't think Safe Mode even lets you install plugins.

#### Version of OctoPrint

1.4.2

#### Operating System running OctoPrint

Raspibian Buster

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Firefox 81.0.2 (32-bit)

#### Link to octoprint.log

[logs.tar.gz](https://github.com/OctoPrint/OctoPrint/files/5396835/logs.tar.gz)

#### Link to contents of terminal tab or serial.log

above link to ALL logs.

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

[image]
[image]

I have read the FAQ.

jneilliii, 2020-10-18:
> Just to clarify something, software update restarting at the end is expected behaviour. It doing it after updating each individual plugins when you click update now for a while bunch to be updated concurrently isn't normal. 

quoting @foosel's comment in Discord the `Update Now` button on the pop up is expected to restart at the end of the install. What is not suppose to happen is for OctoPrint to restart if you use just the `Update` plugin from within the plugin list in the Software Update section of settings. 

I experienced the later of these two circumstances where when I used the `Update` button from within Software Update settings on a single plugin of two that were available to upgrade octoprint restarted.

CmdrCody51, 2020-10-18:
> > Just to clarify something, software update restarting at the end is expected behaviour. It doing it after updating each individual plugins when you click update now for a while bunch to be updated concurrently isn't normal.
> 
> quoting @foosel's comment in Discord the `Update Now` button on the pop up is expected to restart at the end of the install. What is not suppose to happen is for OctoPrint to restart if you use just the `Update` button from within the plugin list in the Software Update section of settings.
> 
> I experienced the later of these two circumstances where when I used the `Update` button from within Software Update settings on a single plugin of two that were available to upgrade octoprint restarted.

I saw that in Discord.

> After discussing this on Discord, I took the next Update and used the Update Now button. When I add to this Issue
> I will use the Plugin Manager.

And I will update when I have more updates I promised I would get an issue up here on the weekend and - here it is.
As I mentioned there - this is more an 'inconvenience' than as a real bug.

foosel, 2020-10-19:
> What is not suppose to happen is for OctoPrint to restart if you use just the `Update` button from within the plugin list in the Software Update section of settings.

That is not 100% correct.

Let me explain how it's supposed to behave, regardless of what you click.

When you initiate an update, OctoPrint will look at the targets it is supposed to update. That can be one in the case of the individual update button, or it can be many in case of an "Update now" button being clicked on a notification for a number of pending updates. 

For each update target OctoPrint will determine if the update requires a) a system reboot, b) an OctoPrint restart, c) a UI reload or d) no further action. The action with the highest impact here wins and will be applied after the update. If you update only one plugin that requires a reload, only a reload will be requested. If you update one plugin that requires a restart, a restart will be done. If you update 20 plugins, 19 of which require no action and one requires a full blown system reboot, a system reboot will be done. It all depends on the plugins that you update, what mixins and hooks they implement and what further update behaviour they have specified in their update configuration.

All of the "update" buttons trigger the exact same processing, the only thing they differ in is the supplied update targets (only one in case of the individual buttons, a list of all pending updates in case of an "update all" button).

This behaviour has not changed ever since the birth of the plugin system and the software update mechanism back in 2015. What HAS changed here and there are what kind of mixins or hooks trigger what kind of post-update action, and obviously plugins also change what mixins and hooks they implement. The post-update action is a crucial part of the update process and MUST be done, otherwise the system is left in a floating incomplete state that could cause all kinds of issues with half applied updates. It cannot be removed from the whole update process without causing trouble.

I do agree that it would make sense to introduce a way to make a selection of all currently pending updates, update those in one go with only ONE restart/reload/whatever, but that would have to be done through a redesign of the current update list UI, allowing check box selection or something like that. That would be more like a feature request however.

foosel, 2020-10-19:
I think I need more information on what exactly happened here with which plugins before I can figure out what to do with this ticket. It would also help if you could share the `plugin_softwareupdate_console.log` from the time you ran multiple updates for which you saw the looping issue, as that should contain any kind of update errors you might have encountered. Alternatively it would be interesting to get it once the issue reproduces. I can't see anything suspicious in the logs you provided.

foosel, 2020-11-02:
@CmdrCody51 anything you could add (e.g. a log)? Otherwise I'll have to close this as not reproducible and not actionable.

closed 2020-11-02 by CmdrCody51