OctoPrint/OctoPrint #3784: Flashforge Creator Pro (2016) stuck in "cancelling" state

DaveTSG, 2020-10-24:
#### What were you doing?

During printing (with a Flashforge Creator Pro, 2016 model), if I cancel or pause any print, the connection status gets stuck in "Cancelling" and I have to power-cycle the printer and then reconnect to it to regain control.

#### What did you expect to happen?

The print to cancel and for Octoprint to retain control over the printer, and not get stuck in the 'cancelling' state, and to be able to start a new print if desired.

#### What happened instead?

Octoprint got stuck in "cancelling" and lost all control over the printer.

#### Did the same happen when running OctoPrint in safe mode?

The FFCP cannot be operated in safe mode as it needs the GPX plugin to run. However, if I disable all plugins except the GPX plugin, the error still occurs.

### What did you already try to solve it?
I found a bunch of earlier threads on this (dating back to 2018):
[This](https://community.octoprint.org/t/canot-cancel-print-in-octoprint/1966)
[This one](https://github.com/OctoPrint/OctoPrint/issues/1139).
[This](https://github.com/OctoPrint/OctoPrint/issues/2581).
And [this one](https://community.octoprint.org/t/octoprint-1-3-7-1-3-8-is-stuck-in-pausing-or-cancelling/1625).... 
They all seem to have been resolved, though, so weren't of any help to me.

I have tried disabling "log position on pause" and "log position on cancel". Neither option fixed the problem.
I also tried cancelling a print and then clicking the "Fake acknowledgement" button, but this seemed to do nothing (nothing appeared in the terminal other than the printer's regular reporting of temperatures).
I tried clicking the "Fake acknowledgement" button a dozen or so times. Eventually a message was sent (which I saw in the terminal), but the connection status remained "cancelling". Tried a few more times and eventually the connection status changed to "operational". However, I was still unable to actually control the printer (I couldn't move the extruders using the X/Y buttons on the Control tab, for instance. I was able to turn on the print bed heater, but I was unable to turn it off.

Each time I get stuck in this situation, I power-cycle the 3D printer and reconnect and that fixes things. But that's not ideal, for obvious reasons...

As per [this](https://community.octoprint.org/t/octoprint-1-3-7-1-3-8-is-stuck-in-pausing-or-cancelling/1625), I tried also manually sending M114, but nothing seemed to be sent (no outgoing message to the printer appeared in the terminal above). I tried this multiple times without success.

I posted about this issue in [a FF Facebook group](https://www.facebook.com/groups/913241999042183/permalink/1302175810148798/?__cft__[0]=AZUL4X9fDU7CAD-kVnJoKQzP2kas4vyOzKaNCqyLcV4nHG17nfqBkVjck5PWblOx00ZzYo7faQDtpCc_lRxriqm108osxtdb-cTO40M_Tbq69A7sdDof7LUbWRnDmOQ15gUkiKQWxAS4d2rrhxXCO-vXAKSx6NXueaF8rWYrek3S4g&__tn__=%2CO%2CP-R) and it was suggested I make the pause and cancel GCODE scripts completely empty. I tried that, but the issue persisted.

#### Version of OctoPrint

OctoPrint version 1.4.2

#### Operating System running OctoPrint

OctoPi v0.17.0

#### Printer model & used firmware incl. version

FlashForge Creator Pro (v1/2016 model)
Printer firmware: Sailfish v7.8.

#### Browser and version of browser, operating system running browser

Tested on Safari, Google Chrome, and Firefox on two different Macs, a Windows computer, and an iPad. Issue doesn't seem to be browser-specific.

#### Link to octoprint.log

My entire logs folder can be found [here](https://community.octoprint.org/uploads/short-url/a0YVCTnTPk1FkZruXDaeP3Fn3zJ.zip).

#### Link to contents of terminal tab or serial.log

My entire logs folder can be found [here](https://community.octoprint.org/uploads/short-url/a0YVCTnTPk1FkZruXDaeP3Fn3zJ.zip).

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ. 

I have also [posted this on the Octoprint forum](https://community.octoprint.org/t/flashforge-creator-pro-2016-stuck-in-cancelling-state/25026) a fortnight ago but got no responses.

Please note that I also upgraded to Python 3 to see if that would fix the issue, but it did not. This upgrade to Python 3 is [described in more detail here](https://community.octoprint.org/t/flashforge-creator-pro-2016-stuck-in-cancelling-state/25026/2?u=davetsg), but the gist of it is the update itself was successful. I also have a serial.log file from when I replicated the error after upgraded to Python 3, [here](https://community.octoprint.org/uploads/short-url/5ykewN2Fe2dAhbnJnLAUnv7yWtl.zip).

Thanks in advance for any help.
Dave

cp2004, 2020-10-25:
Looks like this is an issue with the GPX plugin, rather than OctoPrint. A quick glance at the plugin's issue tracker I spotted https://github.com/markwal/OctoPrint-GPX/issues/77 which shows the same problem your printer does.

The key item that would be most useful here, would be the serial.log. This needs to be enabled, hence in your logs it is not there. That would probably identify if the problem was with OctoPrint or the GPX plugin. 

@markwal said (in the issue I linked above) that he had made some progress towards fixing it, but has since not had time. My money is on this being a problem with the GPX plugin. (https://github.com/markwal/OctoPrint-GPX/issues/77#issuecomment-652647783)

DaveTSG, 2020-10-25:
My bad, [here's a serial.log file](https://www.dropbox.com/s/mrd67oshcj6fn1m/serial%20post%20Py3%20upgrade.log?dl=0). 
Does this log tell you it's a GPX issue? If so I shall follow-up in that linked issue.

Thanks for your help.

foosel, 2020-10-26:
The serial port goes down at the end of that log ("unexpected OSError"). And that's wrapped by GPX. So yeah, that is likely on GPX. In general, if you need GPX to even communicate with your printer, and you have any kind of issues involving said printer communication, there's no way for me to debug or help you really. Report it to Mark, and if he finds a fault in OctoPrint vs his plugin, he'll raise an issue on my end.

closed 2020-10-26 by foosel