OctoPrint/OctoPrint #3788: Hi Gina, I have something to show you and Gaston has told me to submit the request here. Apparently my progress in Dashboard and the progress of the print vary immensely... I am gonna try posting the video.

Jon-So, 2020-10-29:
#### What were you doing?

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

gdombiak, 2020-10-29:
Uploaded video that @Jon-So  shared with me.
[video-1603917097.mp4.zip](https://github.com/OctoPrint/OctoPrint/files/5456529/video-1603917097.mp4.zip)

fieldOfView, 2020-10-29:
For questions or other kind of help or support, please get in touch on the [community forum](https://community.octoprint.org/). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md).

If you think your issue is a problem in the OctoPrint code, and not in an OctoPrint plugin, please fill in the issue template. You did not answer any of the questions in the issue template, making it near impossible for anyone to determine the cause of the issue.

fieldOfView, 2020-10-29:
I'm closing this issue because the issue template is not filled in and the problem in the video seems to be more related to the dashboard plugin. For problems with the dashboard plugin, please post an issue in the issue queue for that plugin: https://github.com/j7126/OctoPrint-Dashboard/issues

closed 2020-10-29 by fieldOfView

Jon-So, 2020-10-29:
But it's inconsistent with Octoprint itself

cp2004, 2020-10-29:
The two things you point to in the video are:
* The dial on the dashboard plugin
* A 'message' which I'm assuming is injected by another plugin, which sometimes shows the right progress, sometimes not.

The progress bar in OctoPrint is not visible in that video, so that's not where this inconsistency is coming from. 

Also, there is no way for people to work out what the problem is since you provided very little information in the report, not filling in any of the template at all.

Jon-So, 2020-10-29:
I showed it to Gaston the maker of the OctoPod app for iPhone and he told me I should post it here because it's a math issue from Octoprint itself, and the message was intended for Gina... I don't mind filling out the template but there is an actual video to point out the differences... I will repost with the template and include dashboard in it if possible and see where it goes... just trying to solve a bug... didn't realize it's like solving world hunger... lol and I am not trying to be difficult but they could just ask me more info instead of just pushing it to the side and closing the ticket...

foosel, 2020-10-29:
I would have told you exactly the same. I cannot analyse bugs without a filled out reporting template and the requested logs. Not because I don't want to but simply because I do not have enough information to do anything but shrug helplessly. In this specific case I would also ask you to provide a gcode file with which to reproduce.

Gaston probably asked you to open a bug report and not just send me a message.

And please don't treat this like we are asking you to do the impossible here. The minimal effort it takes to fill out the report properly and provide the requested information, all of which is really available, is not even remotely comparable to the amount of work running this project is, or how much time gets lost by having to run after logs and such. So please - just provide the needed info, reproduced in safe mode.