OctoPrint/OctoPrint #3789: BUG: Missing Version Number

The009, 2020-10-30:
#### What were you doing?

1. Go to '...' Go to the main interface
2. Click on '....' nothing
3. Scroll down to '....' the bottom
4. See error :     There is no version on octoprint which breaks plugins that require a version number.

 OctoPrint 0+unknown Python 2.7.16 

#### What did you expect to happen?
A version number to be shown
#### What happened instead?
     OctoPrint 0+unknown Python 2.7.16 
#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint

OctoPrint: 0+unknown
Installed: Commit c20c9946f87b1da1216c6af1d1a688036d97a6e7 
Available: Commit c20c9946f87b1da1216c6af1d1a688036d97a6e7

#### Operating System running OctoPrint

Linux version 5.4.72-v7l+ (dom@buildbot) (gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)) #1356 SMP Thu Oct 22 13:57:51 BST 2020

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser
Firefox Nightly - 84.0a1 (2020-10-29) (64-bit)

#### Link to octoprint.log
https://content.the009.net/temp/octoprint.txt

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:
https://content.the009.net/temp/Screenshot_2020-10-29%20OctoPrint.jpg

I have read the FAQ.

jneilliii, 2020-10-30:
That looks as if your octoprint was installed via a git clone or commit? I've only see that when switching to github releases on maintenance branch for development.

The009, 2020-10-30:
It was installed via the master branch using the guide in the readme.md 

"Or alternatively, for an install from source:

    Checkout OctoPrint: git clone https://github.com/OctoPrint/OctoPrint.git
    Change into the OctoPrint folder: cd OctoPrint
    Create a user-owned virtual environment therein: virtualenv venv
    Install OctoPrint into that virtual environment: ./venv/bin/pip install .
"
https://content.the009.net/temp/Screenshot_2020-10-30%200.png

Never had an issue till now

cp2004, 2020-10-30:
Yeah, those settings are the reason for this. When you are using a local git clone, it should be set as 'Local Checkout', rather than GitHub commit. Since downloading the zip per-commit doesn't have the necessary metadata to calculate the version number

Once you switch that update configuration to local checkout, you might need to reinstall using `pip install -e .` to get it to start using the checkout again.

The009, 2020-10-31:
Fair enough, you may want to change the default settings then when someone clones the repo, as I have not gone in there and set that :)

foosel, 2020-11-02:
I don't know what you did there, but when I do a `git clone` followed by a `pip install .` in a venv, I get a version number just fine:

```
$ git clone https://github.com/OctoPrint/OctoPrint OctoPrintInstallTest
Cloning into 'OctoPrintInstallTest'...
remote: Enumerating objects: 79, done.
remote: Counting objects: 100% (79/79), done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 70022 (delta 29), reused 43 (delta 18), pack-reused 69943
Receiving objects: 100% (70022/70022), 42.62 MiB | 17.81 MiB/s, done.
Resolving deltas: 100% (45564/45564), done.

$ cd OctoPrintInstallTest/

$ virtualenv venv
created virtual environment CPython3.8.5.final.0-64 in 818ms
  creator CPython3Windows(dest=D:\Code\OctoPrint\OctoPrintInstallTest\venv, clear=False, global=False)
  seeder FromAppData(download=False, pip=latest, setuptools=latest, wheel=latest, via=copy, app_data_dir=C:\Users\Gina\AppData\Local\pypa\virtualenv\seed-app-data\v1.0.1)
  activators BashActivator,BatchActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator

$ ./venv/Scripts/pip install .
Processing d:\code\octoprint\octoprintinstalltest
[...]
Successfully built OctoPrint tornado netifaces wrapt emoji
Installing collected packages: OctoPrint-FirmwareCheck, OctoPrint-FileCheck, markupsafe, tornado, markdown, pyasn1, rsa, regex, Click, Jinja2, werkzeug, itsdangerous, flask, Flask-Login, pytz, Babel, Flask-Babel, webassets, Flask-Assets, cachelib, PyYAML, pyserial, netaddr, pathtools, watchdog, sarge, netifaces, pylru, pkginfo, urllib3, certifi, idna, chardet, requests, semantic-version, psutil, feedparser, future, six, websocket-client, wrapt, emoji, frozendict, sentry-sdk, filetype, unidecode, blinker, OctoPrint
Successfully installed Babel-2.8.0 Click-7.1.2 Flask-Assets-2.0 Flask-Babel-1.0.0 Flask-Login-0.5.0 Jinja2-2.11.2 OctoPrint-1.4.2 OctoPrint-FileCheck-2020.8.7 OctoPrint-FirmwareCheck-2020.9.23 PyYAML-5.3.1 blinker-1.4 cachelib-0.1.1 certifi-2020.6.20 chardet-3.0.4 emoji-0.6.0 feedparser-5.2.1 filetype-1.0.7 flask-1.1.2 frozendict-1.2 future-0.18.2 idna-2.10 itsdangerous-1.1.0 markdown-3.1.1 markupsafe-1.1.1 netaddr-0.8.0 netifaces-0.10.9 pathtools-0.1.2 pkginfo-1.6.1 psutil-5.7.3 pyasn1-0.4.8 pylru-1.2.0 pyserial-3.4 pytz-2020.4 regex-2020.10.28 requests-2.24.0 rsa-4.0 sarge-0.1.5.post0 semantic-version-2.8.5 sentry-sdk-0.19.1 six-1.15.0 tornado-5.1.1 unidecode-0.4.21 urllib3-1.25.11 watchdog-0.10.3 webassets-2.0 websocket-client-0.57.0 werkzeug-1.0.1 wrapt-1.12.1

$ ./venv/Scripts/octoprint.exe --version
octoprint, version 1.4.2
```

It has nothing to do whether you install from a git clone or not, the only scenario where you might run into version detection issues is if you install a tarball from a branch directly fetched from github. Whatever is going on here, it's isolated to your checkout from the looks of it.

Please provide the *exact* commands you used.

The009, 2020-11-02:
When I installed Octoprint a while back I used the clone feature.

After it installed it was automatically checking the master commit on GitHub for updates. 
It updated a few times from the Master Branch on GitHub without fail.

After the last update is when the version number broke. I did not run any commands it did it on its own.
I have the logs cleared on reboot of the device to save space. so I did not have any old logs that would have the many updates.

After I found the bug I did a 
`git pull`
to update my local repo then did a 
`pip install -e .`

this in return did not fix the issue. 
I am sure if I nuke my install do a clean clone then install, then change the settings to local for updates the problem would be gone.

So guess the thing to do would be NOT allow tracking for updates of the GitHub master branch or change the install instructions to include needing to change the update location to local repo.

foosel, 2020-11-02:
>  I totally forgot to do `./venv/bin/pip install -e .` and just did the `pip install -e`

That would have installed it globally which is not recommended because it can cause all kinds of issues thanks to dependency hell.

> So I guess now its just a matter of putting int he docs if you clone the repo you need to change the update settings.

I'm still unclear to what update configuration you have set. It offering commits indicates that you HAD it configured to track a local checkout, but maybe not? What does it say here:

[image]

> So I guess now its just a matter of putting int he docs if you clone the repo you need to change the update settings.

It will already tell you that quite prominently right in the update plugin:

[image]

In general though, installing from git is really only meant for development purposes and usually you should stick to releases instead. You don't really gain anything from git cloning when you run the `master` branch.

The009, 2020-11-02:
I have been doing dev work :) that is why i had to change my local branch to master before doing the pull.

When I first installed it (months ago I think this copy was done around May), the settings were set automatically to Github commit. Because it was set automatically to commit and the master GitHub branch I have just always let it update that way.
That is when this current issue happened. 

Then once I made this I was told to change it to Local checkout,

If I set it to stable I get the following message `Make sure you have read "How to use the release channels to help test release candidates" before switching to a release channel other than "Stable"`
with Version Information of
 `OctoPrint: 1.4.2
Installed: 1.4.2 
Available: 1.4.2 (stable release) Release Notes: https://github.com/OctoPrint/OctoPrint/releases/tag/1.4.2`

If I change it back to Github commit I get
`OctoPrint: 1.4.2
Installed: Commit c20c9946f87b1da1216c6af1d1a688036d97a6e7 
Available: Commit de29cdb7e568150df1e452f67f12ab53fd908436`

But If I change it to Local checkout I and I set the OctoPrnt checkout folder to `/home/pi/Octoprint` I get
`OctoPrint: 1.4.2
Installed: ? 
Available: ?`

With a ! with a tooltip of "Update not possible"

I hope this info help, you have been incredible helpful and kind sofar.

Mine never said that . It may have been a plugin I added that may have blocked that message or maybe a something in my browser.

I have the setting

foosel, 2020-11-02:
> /home/pi/Octoprint

That should probably say `/home/pi/OctoPrint`, unless you've manually cloned into a separate folder. Linux file systems are case sensitive and `/home/pi/Octoprint` is not the same as `/home/pi/OctoPrint`. When you do a `git clone` it will automatically created `OctoPrint`, not `Octoprint`, unless specified otherwise. Try changing that and see if that fixes things. Your current ? reports indicate that it can't read the checkout, which would make sense with a wrong path.

> the settings were set automatically to Github commit

That... sounds wrong. OctoPrint ships with release tracking by default for several years now, anything else you'll have to configure yourself. If you only pulled in May, this doesn't make sense.

The009, 2020-11-02:
Well that is embarrassing, you are correct. the config.yml was stored from my last dev run and I did not even think of that. totally totally my fault. Sorry. and the Case on Nix vs Windows was the issue. Thank you.
Doing dev on Windows and Testing on the PI with a printer won't be an issue with Case now.

But If I am thinking correctly this may happen again correct?
If I am doing pull requests to make sure that my dev work keeps up with the dev work that you are all doing the update may break the version number after a pull then pressing update.

If that happens again to correct the issue I should just run the `./venv/bin/pip install -e .`

foosel, 2020-11-02:
> If I am doing pull requests to make sure that my dev work keeps up with the dev work that you are all doing the update may break the version number after a pull then pressing update.

If you are actively working on the source then don't update through OctoPrint but do it manually from the command line. OctoPrint will not install in editable mode (the "-e" part of the pip install there) and that will lead to what is installed and what is checked out diverging. If you intend to develop on that checkout, keep it updated manually. Otherwise you are fine to keep it updated through Local Checkout tracking (NOT Github Commit tracking). In general, for development you really rather should go with a local install on your development machine instead of mucking around on a Pi, it will make your life so much easier.

Anyhow... closing this since - as far as I understand - the issue has cleared itself up now after proper install and configuration.

closed 2020-11-02 by foosel

The009, 2020-11-02:
> If you are actively working on the source then don't update through OctoPrint but do it manually from the command line. OctoPrint will not install in editable mode (the "-e" part of the pip install [...]