OctoPrint/OctoPrint #3790: "Uncaught TypeError: pipArgs is null" When Setting "Force the use of the --user flag with pip install"

poita66, 2020-10-30:
#### What were you doing?

Enabling the "Force the use of the --user flag with pip install" setting

1. Go to Settings
2. Click on Plugin Manager
3. Click on spanner icon to open Plugin Manager Settings
4. Check "Force the use of the --user flag with pip install"
5. Click Save
6. Open dev tools
7. See error "Uncaught TypeError: pipArgs is null"

#### What did you expect to happen?

The setting to update and the UI to confirm that the save was successful

#### What happened instead?

After pressing save, nothing

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

1.4.2

#### Operating System running OctoPrint

Linux (Fedora 32 Server)
Python 3.8.5

#### Printer model & used firmware incl. version

N/A

#### Browser and version of browser, operating system running browser

Firefox 82.0 on Fedora Silverblue 33

#### Link to octoprint.log

https://gist.github.com/poita66/5a8a5d37fac203619713a7adf9fb7a14

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

https://gist.github.com/poita66/ab42f9d2d9348cb4005d8158b606c13d

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2020-10-30:
Confirmed bug in 1.4.2, but has already fixed for next release by https://github.com/OctoPrint/OctoPrint/commit/6ea2caa770c573115d75317505c2a68d3cd35101 already, but no ticket reporting this yet.

foosel, 2020-11-30:
1.5.0 has just been released.

closed 2020-11-30 by foosel