OctoPrint/OctoPrint #3791: [Request] Able to restrict interfaces it listens or SSDP broadcasts

krisavi, 2020-11-01:
**Is your feature request related to a problem? Please describe.**
Yes, it is due to a problem. It kept bombing virtual machine and docker network with the discovery plugin. At first tried to set the "host" parameter in config -> server section. It did not restrict it.
This might not be that needed on OctoPi solution, but more relevant on homeserver setups where OctoPrint is just one service.

**Describe the solution you'd like**
One of the possible solutions would be to have user made interfaces override, like in config to be to set up the interfaces that Octoprint could bind to and use for plugins. 

**Describe alternatives you've considered**
I considered disabling all port 1900 requests in whole home network, just because I did not like to see lines about it in firewall log. I know I could easily only restrict it to specific interface using firewall, but I find the application itself should be able to limit the scope. 

**Additional context**
Ended up modifying https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/util/net.py#L138 into following:
```python
    from octoprint.settings import settings
     interface_override = settings().get(['server', 'interface_override'])
     if interface_override:
         interfaces = settings().get(['server', 'interfaces'])
     else:
         interfaces = netifaces.interfaces()
     for interface in interfaces:
```
and adding in config.yaml:
```yaml
server:
    interface_override: true
    interfaces:
    - eth1
```
That does what I need. It does not try to broadcast to ipv6 tunnel, vpn network, incoming ppp connection nor VM and docker networks. That way I set up discovery in scope where I can actually physically access the printer. 
It is still listening on those networks, but at least the discovery broadcasting stopped.

foosel, 2020-11-02:
Implemented by the above commit, to be shipped as part of 1.5.0.

I went a step further and also added an option to specify addresses on which to enable discovery (not just interfaces), and restricted SSDP responses to the enabled scope as well.

Configuration is plugin specific however, so not `server.interfaces` but rather `plugins.discovery.addresses` and `plugins.discovery.interfaces`. No need for a boolean flag, if the list is provided it gets used, otherwise the old default behaviour applies. See also https://docs.octoprint.org/en/maintenance/bundledplugins/discovery.html#configuring-the-plugin, which has been adjusted accordingly.

krisavi, 2020-11-02:
I did notice after I made a workaround in my environment that the boolean was not needed as I could test the existence of the variable and that it might affect more than the plugin I wanted to quiet, so the new solution is more correct. 
I noticed the listen address for Octoprint is defined by server.host paramater, so that is working fine as well and net.utils interface_addresses function does not affect it. 
The ability to split between addresses and interfaces is quite nice as well.

foosel, 2020-11-30:
1.5.0 has just been released.

closed 2020-11-30 by foosel