OctoPrint/OctoPrint #3792: Code style is not followed.

krisavi, 2020-11-01:
#### What were you doing?
When I was making my own workaround for the issue described in #3791 I found it quite annoying that the core files were indented using tab. I did check the https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md which states the code style in pull requests sections.
```
7. Make sure you follow the current coding style. This means:

Spaces for indenting and alignment, indentation width 4.
English language (code, variables, comments, ...)
Comments where necessary: Tell why the code does something like it does it, structure your code
Following the general architecture
If your PR needs to make changes to the Stylesheets, change the .less files from which the CSS is compiled.
Make sure you do not add dead code (e.g. commented out left-overs from experiments).
```
The files like https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/util/net.py are using tab instead of spaces for indenting. 

Edited the files in vim, got warnings when tried to save and error when trying to run application.

#### What did you expect to happen?
No warnings and errors.

#### What happened instead?
After editing file it gave syntax error. In vim when, it made indentation using spaces and width of 4. Rest of the code was with tabs and it clearly is a problem for python to mix indentation styles as it uses indentation as part of code structuring and syntax. Since there is also no vim modeline in files to override default, then it assumed the default python configuration, where indentation of 4 spaces is used.

#### Did the same happen when running OctoPrint in safe mode?

I am quite sure it would have resulted in same issue.

#### Version of OctoPrint

OctoPrint 1.4.2

#### Operating System running OctoPrint

Debian 10.6

#### Link to octoprint.log

Irrelevant, the issue is due to workaround modification being indented according to more default for python, that is also described in PEP8 did not match with the existing coding style.

I have read the FAQ and about the code style that seems to be agreed.

jneilliii, 2020-11-01:
Were you using the maintenance branch to make your changes? I think the editorconfig has been updated there for the space indenting.

cp2004, 2020-11-01:
Development is done against `devel` or `maintenance`, both of which have new pre-commit hooks that include all of these code formatting changes. PRs should target those branches, so it is important it is documented under 'PRs'. However, this hasn't made it's way into the latest release which is on master.

Changes were introduced by #3762 and are available on both `devel` and `maintenance`, hopefully that explains it 🙂

closed 2020-11-01 by cp2004

krisavi, 2020-11-01:
Ok, got it. 
Since I did modifications ins site packages, then that one probably has the spaces to tabs done on it. 
I do not see the point of changing to tabs, the editorconfig does not matter on the "released" version. 
Just to experiment with the existing code and to make change seems to be quite inconvenienced just because of the hook that changes python files from spaces to tabs.

foosel, 2020-11-01:
The code has been changed from tabs to spaces and pep8 compliance after 1.4.2 was released (which is still the state of the application code in the `master` branch). If you work against the pointed out `maintenance` branch you should find everything as described in the PR related docs, as already pointed out.