OctoPrint/OctoPrint #3794: Raspberry Pi4 2GB freezes when USB inserted

er9xuser, 2020-11-06:
Hello.

I bought RPi 4 2GB installed 2020-11-01_2020-08-20-octopi-buster-armhf-lite-0.18.0.zip. System boots up no problem. Then sudo apt update...then sudo raspi-config. After all that set wifi connection and connect over ssh and open website. Next i insert usb printer into usb 2.0 or usb 3.0. After about a 30 seconds...sistem freezes and if i reboot...rpi boots up again but can't connect over ssh or http/s. reason TimedOut. If i disconnect usb and reboot system boots up with no problem and ssh and website works. Then again...inserted usb and again freeze. I have powersupply with 5.0/51/5.2 volts and up to 4A. 

what seems to be a problem?

Thank you.

fieldOfView, 2020-11-06:
This is unlikely a bug in OctoPrint itself. It seem more related to a hardware issue. Personally, I would suspect a power issue. Just the metric "5V 4A" does not say anything about the quality of the power supply, or its ability to handle spikes caused by possible bad behavior of the device(s) connected to the Raspberry Pi. Cabling could also be an issue (either between the PSU and the Pi or between the Pi and the printer).

For questions or other kind of help or support, please get in touch on the [community forum](https://community.octoprint.org/). The bug tracker is reserved for bugs, feature requests and brainstorming tickets. See also the [Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md).

closed 2020-11-06 by fieldOfView

er9xuser, 2020-12-08:
Maybe. Thanks