OctoPrint/OctoPrint #3796: Disabling Webcam via Additional permissions not working

charredchar, 2020-11-07:
#### What were you doing?

I am attempting to create a basic "Kiosk" user that can see the dashboard plugin, do some simple tasks like start or cancel prints. This user will run on an old tablet or maybe even a screen attached to the Pi that is driving the printer so I do not need a webcam, I'd prefer it not enabled to be lighter on resources. I don't select the "Webcam" option under "Additional permissions" for the user but it still loads the webcam stream for that user when they logged in. I don't want to disable the webcam entirely as I'd like to see it when away from the printer.

#### What did you expect to happen?

With the Webcam permissions unselected I expect it not to load under that user.

#### What happened instead?

When loading the Control tab or enabling webcam for Dashboard the webcam window still streams the webcam.

#### Did the same happen when running OctoPrint in safe mode?

Yes, I even attempted to disable single plugins one at a time to ensure it wasn't one of them.

#### Version of OctoPrint

OctoPrint 1.4.2, Python 2.7.16

#### Operating System running OctoPrint

OctoPi 0.17.0

#### Printer model & used firmware incl. version

CR-10 Mini, though I wouldn't think it was related.

#### Browser and version of browser, operating system running browser

Multiple including Chrome Version 86.0.4240.111 on Windows and Chromium on Pi and Android.

#### Link to octoprint.log

https://pastebin.com/EqHsZ9n7

#### Link to contents of terminal tab or serial.log

I didn't have this enabled at the time so nothing is in the log... Considering it is the webcam and permission issue would this really be needed?

#### Link to contents of Javascript console in the browser

https://pastebin.com/XeFG0PzV

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

jneilliii, 2020-11-07:
I think this might require some updates on the dashboard plugin as well, depending on how they are including the webcam on their interface, but I can confirm that even creating a custom group without webcam enabled for that group and assigning to a user the webcam is still loaded/visible.

foosel, 2020-11-07:
As @jneilliii pointed out, this is on the plugin to correct. OctoPrint has no control over the stream itself. What the permission does is controlling whether it gets displayed or not. If a plugin displays the stream without checking for the permission, that needs to be solved in the plugin.

charredchar, 2020-11-07:
> As @jneilliii pointed out, this is on the plugin to correct. OctoPrint has no control over the stream itself. What the permission does is controlling whether it gets displayed or not. If a plugin displays the stream without checking for the permission, that needs to be solved in the plugin.

The plugin is irrelevant in this case, I only pointed it out because it displays the stream on the first tab but even so the Control tab still shows the stream in Safe Mode.

[image]

The issue is the permission does not work and it displays the webcam regardless, if the permission is corrected then it wouldn't show up in Dashboard anyway.

cp2004, 2020-11-07:
Looks to me like the webcam area has the section:

```html
<div [...] data-bind="visible: loginState.hasAnyPermissionKo(access.permissions.WEBCAM, access.permissions.CONTROL)">
```
So it would show if it had permission of EITHER webcam or control.

Confirmed reproduced:
* have test user, 
* add no permission groups, 
* all individual permissions *except* for webcam, 
* still shows webcam in tab

jneilliii, 2020-11-07:
I attempted a fix for this, but for some reason the ko binding didn't work. I wrapped both webcam containers with `` but it still ends up loading for some reason.

jneilliii, 2020-11-07:
Above PR should fix this.

charredchar, 2020-11-07:
> Looks to me like the webcam area has the section:
> 
> ```
> <div [...] data-bind="visible: loginState.hasAnyPermissionKo(access.permissions.WEBCAM, access.permissions.CONTROL)">
> ```
> 
> So it would show if it had permission of EITHER webcam or control.
> 
> Confirmed reproduced:
> 
> * have test user,
> * add no permission groups,
> * all individual permissions _except_ for webcam,
> * still shows webcam in tab

You're awesome! Guess since I still want to be able to control the printer from the Kiosk this kept the camera stream enabled but since I am physically at the printer I didn't need the stream Thanks!

jneilliii, 2020-11-07:
Basically it will either render the webcam image area or not based on if the logged in user has webcam permissions in access control with my submitted changes. This means it will work the way I think it was intentionally meant to work when granular permissions were added to OctoPrint.

charredchar, 2020-11-08:
Since I am using the OctoPi image is there an "easy" way to merge the pull request with my version or should I just wait until it gets merged with an update? I haven't spent enough time with OctoPrint yet to know how long fixes take to get pushed out.

jneilliii, 2020-11-08:
You can git clone the repo local and install from source. There's a small chance that this will get merged in for 1.5.0 release candidate phases, which I believe to be starting sometime next week. If it does get merged to that release then you could just change your release channel in software update settings to maintenance RCs, and you'll get notified when it's out and available and help make sure that the next stable release version is bug free.

https://community.octoprint.org/t/how-to-use-the-release-channels-to-help-test-release-candidates/402

foosel, 2020-11-09:
Fixed by the mentioned PR, will still make it into 1.5.0rc1 👍

charredchar, 2020-11-09:
> Fixed by the mentioned PR, will still make it into 1.5.0rc1 👍

Great to hear! I actually just changed my release channel to Maintenance finding out about this, just waiting for OctoPrint to trigger an update.

charredchar, 2020-11-11:
So I was able to get 1.5.0rc1 installed today and it still shows the webcam in my "Kiosk" user. :slightly_frowning_face: I tested in Safe Mode as well.

jneilliii, 2020-11-11:
I think we're going to need more details to how you have your "kiosk" mode configured. I just tested 1.5.0rc1 adding a user named test with the following access permissions, and no group assignment and when I log in as that user and go to the control tab the webcam does not load.

[image]

[image]

charredchar, 2020-11-22:
> I think we're going to need more details to how you have your "kiosk" mode configured. I just tested 1.5.0rc1 adding a user named test with the following access permissions, and no group assignment and when I log in as that user and go to the control tab the webcam does not load.

I am sorry, I lost track of this! I forgot that I added a group to that user after I couldn't get the webcam to not load and never took it off. It works now! Works great, thank you so much.

closed 2020-11-22 by charredchar