OctoPrint/OctoPrint #3799: [Request] Include full jQuery UI module to avoid conflicts by plugins

jneilliii, 2020-11-07:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when another plugin causes a conflict with one of my plugins because of jQuery UI modules that have been added in order to have the additional functionality of those modules that are included by OctoPrint by default.

**Describe the solution you'd like**
Include the full jQuery UI module with all options so that plugin authors don't have to include their own individual bits and pieces.

**Describe alternatives you've considered**
Not sure if there is another way to get around this outside of possibly scoping. Not really familiar with how to do this and if it's even possible.

**Additional context**
https://github.com/jneilliii/OctoPrint-ConsolidatedTabs/issues/32

foosel, 2020-11-12:
Hm... that's a bit of a tricky one. If I could I frankly would throw it out altogether, the only reason it's (partially) included is due to the jquery file upload dependency needing it. And if I were to bundle it now that would then only cause issues for plugins that include their own versions, wouldn't it?

I fear this is a problem not easily solved...

jneilliii, 2020-11-12:
I don't know if it would cause issues, because if the plugins are already bundling it, their implementation would still overwrite the existing one, depending on load order of when it's bundled, or added to the page if not bundled? The idea is that if the full one was included plugin's wouldn't have to include their own. Just thinking in regard to the number of plugins that I have bundled a full UI module the more plugins you install the more the bloat increases. If I didn't have to bundle, then the combined size of multiple plugin installs would get reduced drastically. 

I'm more than happy to try and figure out how to alias jquery and the jquerui within my own plugins (any guidance/links appreciated), but then the bloat would still potentially be there.