OctoPrint/OctoPrint #3801: Plugin list and Software update list extends beyond the configuration window

dewi-ny-je, 2020-11-09:
I'm setting up Octoprint from the configuration window

Windows 10 resolution: 1600x900, 125% zoom 

1. Go to Plugin Manager or Software Update
2. notice the double scrollbar

#### What did you expect to happen?
There should be only a scrollbar: either the "inner" list is resized to the visible area of the configuration dialog, or the inner list (the plugins) is expanded completely

#### What happened instead?

the inner list has a fixed size which is in my case bigger than the Octopring modal configuration window, resulting in more difficult scrolling, especially in laptops (which are the computer most likely to show the problem due to the smaller resolution.

#### Did the same happen when running OctoPrint in safe mode?

yes

#### Version of OctoPrint
1.4.2

#### Operating System running OctoPrint
Windows 10 20H2

#### Printer model & used firmware incl. version
Not relevant.

#### Browser and version of browser, operating system running browser
Chrome 86

#### Link to octoprint.log
[image]
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/5513423/octoprint.log)

#### Link to contents of terminal tab or serial.log
not relevant

#### Link to contents of Javascript console in the browser
https://pastebin.com/uNCPz3fW

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

cp2004, 2020-11-27:
PRs #3835 and #3836 have been created to address this. Covering both plugin manager, and SW update tabs.

Final one is the 'Get More' tab of the plugin manager, that double scrolls too:
[image]

My initial idea for this was to make it a larger modal (same width as settings), then divide in two, with repository on the left, and URL and file upload on the right. This idea was also with (partial) implementation of #3763 in mind, to create a full export and import feature following #3711 implementation, where the import from the json file was an extra field.

foosel, 2021-04-27:
1.6.0 has just been released.

closed 2021-04-27 by foosel