OctoPrint/OctoPrint #3802: [RC Feedback] Feedback on the 1.5.0rc4 Release Candidate

foosel, 2020-11-10:
Please provide *general* feedback on your experience with the 1.5.0rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

### If you run into any obvious bugs not yet listed below the following line, please *open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports, not just a statement that something doesn't work!

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

schnello, 2020-11-10:
Installed at my pi with Klipper FW. System is up and running. Print test in progress

jneilliii, 2020-11-10:
Maybe we should start tracking a list of broken plugins as a result of the upgrade? So far, prettygcode as you mentioned, bed level visualizer, widescreen, Tasmota, WemoSwitch, TerminalCommandsExtended, TasmotaMQTT, ConsolidatedTabs, MQTTSubscribe, ConsolidateTempControl.

cp2004, 2020-11-10:
Pleased to report my plugins seem to work, WS281x LED Status, Virtual Printer Settings, Marlin EEPROM editor and the bare minimum that is WLED.

foosel, 2020-11-10:
@jneilliii I've added a list to the OP

jneilliii, 2020-11-10:
add Tab Order to that list...

jneilliii, 2020-11-10:
I have release candidates available for all plugins except Bed Level Visualizer that will fix any known conflicts with OctoPrint 1.5.0. The Bed Level Visualizer one is going to take me a while to dig into the code that was added by another contributor to figure out why `$parentContext.$index()` doesn't work in this new knockout version.

foosel, 2020-11-10:
Updated (I hope I got that correctly)

Crowlord, 2020-11-11:
Apart from the already identified plugins all working fine on my single test instance. 

Anycubic Kossel LP (Stock firmware)
Pi3B+
Octopi 0.17
Python 3.7.3

cp2004, 2020-11-11:
3rd party plugin issue: Action Command Prompt
#3810 - fix has been provided to the plugin author: https://github.com/benlye/OctoPrint-ActionCommandsPlugin/pull/5

benlye, 2020-11-11:
> 3rd party plugin issue: Action Command Prompt
> #3810 - fix has been provided to the plugin author: [benlye/OctoPrint-ActionCommandsPlugin#5](https://github.com/benlye/OctoPrint-ActionCommandsPlugin/pull/5)

Testing now - will merge shortly.  Thanks!

benlye, 2020-11-11:
Firmware Updater plugin seems to be working fine.

b-morgan, 2020-11-13:
In anticipation of upgrading to 1.5.0rc1, I decided to convert my 1.4.2 from access control disabled to access control enabled. If I failed miserably, I could revert back to disabled and still print! I followed the distructions on [https://docs.octoprint.org/en/master/features/accesscontrol.html#rerunning-the-wizard](url) successfully and created a user. Sure enough, I need to enter a username and password every time I want to use my printer.

In [OctoPrint on Air #35](https://www.youtube.com/watch?v=AuF2gl_EUQo&feature=youtu.be) at approximately 5:30 Gina mentions config files and auto login local and then moves along with the rest of the video. Unfortunately, that wasn't quite enough of a hint as the only obvious config file is config.yaml which I edited to get to this point, and the Access Control section of the documentation didn't mention auto login local at all.

With some help from the @jneilliii on Discord, I was pointed to [https://docs.octoprint.org/en/master/configuration/config_yaml.html#access-control](url) and found that I needed to add "autologinLocal: true", "autologinAs:", and "localNetworks:".

I also saw documentation for "enabled:" but when I added the above to my config.yaml, the "enabled: true" I had added previously had disappeared but there was an undocumented "salt:" in its place.

IMO, the documentation should be updated to include a mention of the auto login stuff on the Access Control page and the config.yaml documentation should include something about salt: and that it replaces "enabled: true". At the very least, this should be documented in the release notes.

foosel, 2020-11-13:
I'm just at this very moment sitting on a guide on how to set up autologin. The `salt` doesn't replace `enabled` (which has been removed completely, the docs have just not yet been updated but will soon) but rather is where OctoPrint stores the salt used for password storage. I so far saw no need to document that, as it is not something users should touch.

jneilliii, 2020-11-13:
Bed Level Visualizer: RC available

b-morgan, 2020-11-13:
I'd add "salt:" to the config.yaml documentation with the text "Used for password hashing, DO NOT TOUCH" or something similar. Acknowledging its existence is all that is needed, IMO.

b-morgan, 2020-11-13:
Now I'm off to backup my existing 1.4.2 SD card so I can replace it with 0.18.0 RC and 1.5.0rc1 and see how much trouble I can get into :smile:

foosel, 2020-11-13:
https://community.octoprint.org/t/how-to-set-up-octoprint-to-autologin-a-single-user-when-connecting-from-the-internal-network/26235

b-morgan, 2020-11-13:
I've upgraded to OctoPi 0.18.0 (32 bit for RPi 3B) and upgraded OctoPrint to 1.5.0rc1. I have the old SD card mounted in a USB SD card reader. The basics are working but I haven't tried printing anything yet. Oh the joys of a fresh OS!

Guilouz, 2020-11-13:
Just try to upload a language pack on two raspberry pi with latest 1.5.0rc1 and nothing happens, impossible to upload it, there is no error.

jneilliii, 2020-11-14:
I can confirm @Guilouz's error. Will post issue if it hasn't already been.

ChrisHeerschap, 2020-11-14:
Updated on three raspi systems, a 4B, a 3B, and a 2B - looks to be working on all of them but I've only submitted one print on the 3B (Prusa Mini) and none yet on the 2B (Anycubic). Several prints on the 4B (to MK3S) and no issues but did discover that the Fullscreen Plugin doesn't seem to be working. I can submit an issue to that project when I get a chance. The Exclude Region plugin was inadvertently tested with a single part failing of a 6 part print and it worked like a champ. 

Only other issue encountered was the change in auth broke my monitoring. I was hitting the `/` URI and looking for "OctoPrint" but now it's a 302. I curled to get that 302 target, loaded the new `/login/?redirect...` URI and my monitoring is happy again.

Will continue to test, but now a QA person so don't usually do much more than normal workflow. If anyone has a suggested procedure for testing everything out, I'm all ears.

oliof, 2020-11-14:
1.5.0rc1 running on an OctoPi 0.17 release with all packages up-todate.

Had an issue with Octoprint web service not running (pi was still up, I got the page telling me how to check whether the service is still runing) where restarting the service failed. The logs were empty, as well as the journal for the octoprint service.

Starting the octoprint service manually,  I got an error message that config.yaml is malformed. Indeed, in the beginning of the config.yaml was some binary data. I deleted everything until valid yaml (`accessControl:...`). After that, the service still didn't come up, but I was able to run octoprint manually now in a tmux session. I started a print, but after about 30min I got a segmentation fault mid-print. Nothing in the logs though.

Attached is the malformed yaml file with authentication data removed.

[config.yaml.broken.gz](https://github.com/OctoPrint/OctoPrint/files/5540917/config.yaml.broken.gz)

foosel, 2020-11-16:
@oliof Huh... that's a new one. That much garbage in the data, combined with a service refusing to start when manual start work, and a segfault to boot - frankly, that sound like something might be wonky with the file system. Unsatisfying explanation, I know, but running a file system check on the card might yield some results. Sadly nothing that I'd have any idea about how to analyse from my side right now, all the update tests that I ran ran through without data corruption (that I know of) and roll-out also doesn't indicate a general issue.

Speaking of which, current stats for 1.5.0rc*:

[image]

Currently targeting a second RC tomorrow or Wednesday the latest (I have some appointments tomorrow that I'll have to work around, hence no full commitment for tomorrow).

ManuelMcLure, 2020-11-16:
Other than Fullscreen Webcam plugin not working, I'm not seeing any issues on 1.15rc1 installed on Debian Buster on an Intel J1900 mini-PC (64 bit). I've been looking into the sources for the Fullscreen Webcam plugin to see if I can figure out what the problem is, but my CSS/JS knowledge is very limited.

JohnOCFII, 2020-11-17:
Preparing to test rc1, and ran into this tiny issue regarding avconv/ffmpeg:  #3821 

[1.5.0rc1] Path to FFMPEG: /usr/bin/avconf missing #3821

foosel, 2020-11-17:
@JohnOCFII That's an Octo**Pi** issue, I've copied your ticket over there.

Fullscreen webcam plugin issue has been added to known third party issues.

foosel, 2020-11-17:
1.5.0rc2 is out, new feedback ticket is here: #3823

closed 2020-11-17 by foosel