OctoPrint/OctoPrint #3803: Restore disabling Access Control through low-level configuration options

ucirello, 2020-11-10:
Adding username and password to OctoPrint is going to break my workflow. I use it inside a VPN, closed from the public. I understand the concern of people exposing their OctoPrint installations, but the same time, this more advanced workflow should not be disturbed. 

My suggestion is to restore this functionality behind a low-level configuration flag. Somewhere deep in the configuration files that would disable it, therefore making it hard for people to unintentionally expose their OctoPrint instances.

cp2004, 2020-11-10:
Are any of the options in `config.yaml` of any use to you?  Maybe, the Auto Login Local (using VPN), or the basic auth headers? https://docs.octoprint.org/en/master/configuration/config_yaml.html#access-control

ucirello, 2020-11-10:
@cp2004 -- I am going to double check if a combination of `autologinLocal`, `autologinAs`, and `localNetworks` works for me and I get back to you in one day.

jneilliii, 2020-11-10:
You know there's also that little checkbox for remember me on the login page as well...minor inconvenience in the grand scheme of things compared to the headaches caused by people that are responsible like you and are using a VPN solution.

ucirello, 2020-11-10:
@jneilliii going through a login screen is not an option for me.

foosel, 2020-11-10:
> @jneilliii going through a login screen is not an option for me.

May I ask why?

ucirello, 2020-11-10:
@foosel -- I am not in position to share details of my operation. Suffice to say that going through a login screen is not an option for me. The non-obvious alternative is use some kind of browser extension to drive the login for me - with all the consequences that it entails (like when OctoPi network fails and you have to reload the screen)

foosel, 2020-11-10:
Well, you have `autologinAs` and automatic login based on provided basic auth headers at your disposal.

I will not bring back the option to run without access control enabled as experience has shown that within weeks someone will just make a tutorial on how to set this low-level configuration option and then I'm back to square one with regards of the constant overhead insecure setups like that exposed on the public internet cause me.

I'm happy to evaluate other options that go beyond either of the aforementioned two autologin options, but allowing to disable access control is not something I will bring back.

ucirello, 2020-11-10:
> I'm happy to evaluate other options that go beyond either of the aforementioned two autologin options, but allowing to disable access control is not something I will bring back.

Agreed. 

As stated in https://github.com/OctoPrint/OctoPrint/issues/3803#issuecomment-724773880 - I get back to you in one day.

ucirello, 2020-11-11:
@foosel -- It didn't do the trick - but for a different reason. There's no automatic setup/wizard for when I move `enabled: false` to `enabled: true` in the version I am running.

cp2004, 2020-11-11:
You *shouldn't* change enabled to true, without having created a user first. OctoPrint 1.5.0rc1+ will have prompted you automatically to enable it - and that configuration option is ignored if it is set to false. This is the basis of my ticket #3804, to better handle when no users are configured but access control is enabled.

ucirello, 2020-11-11:
@foosel @cp2004 -- I followed these steps: https://docs.octoprint.org/en/master/features/accesscontrol.html#rerunning-the-wizard
I managed to get the user created after re-running the wizard. After I got over this bump, the autoLogin features behave like I needed. Thanks. 

@cp2004 -- makes sense. I think I slip through the cracks by misunderstanding the config.yaml documentation. In any case, I can see a way to keep my workflow. Thanks.

closed 2020-11-11 by ucirello

muchsimpler3d, 2020-12-16:
I too, interact with a large number of OctoPrint instances exclusively through a VPN, and use of the Puppeteer Browser interaction library. Like the person who reported above, this change caused MASSIVE problems for me, and I consider this decision to be a truly terrible, user-hostile action by the author of OctoPrint.

Like @cp2004 above, after several hours of research once I realized what was causing all the problems, I found that yes, I could make my setup work with autologinAs and autologinLocal... except that this solution wasn't actually documented anywhere.

So that others may not suffer quite as badly as I've suffered, here is that solution.

Edit the config.yaml file, remove the "enabled: false" line from the accessControl section if it is present and add these three lines to the accessControl section instead (where 'admin' is your admin user):
```
accessControl:
    autologinAs: admin
    autologinLocal: true
    localNetworks:
    - 0.0.0.0/0
```

This will grant completely unrestricted admin access to absolutely everyone. Presto, zero security and zero pesky password prompts. Problem solved!

cp2004, 2020-12-16:
I appreciate you may not like this decision, but you should understand that it was a necessary one - it was causing way to much overhead to field all the security concerns & support people in the community who were upset when their own stupidity of leaving it unsecured on the public internet meant it was discovered. Watch the latest 'OctoPrint on Air' video (36) for a full explanation.

As for the suggestion it is not documented, have you read any of the documentation? There's a guide on the forums, linked from the release notes, the blog post, or if you'd *asked* we could have sent the link if you couldn't find it yourself. There's even a plugin that will help you configure it.

muchsimpler3d, 2020-12-16:
Sure, let's talk about the documentation. 

When I googled "OctoPrint Disable Access Control" the page that came up first was, unsurprisingly, the official OctoPrint documentation for Access Control, which is here: https://docs.octoprint.org/en/master/features/accesscontrol.html

The quote that stands out is:  "Upon first start a configuration wizard is provided which allows configuration of the first administrator account or alternatively disabling Access Control (which is NOT recommended for systems that are directly accessible via the internet or other untrusted networks!)."

In other words, your access control documentation has not been updated to reflect that you can no longer turn off access control.

Oh, but there are the 1.5 release notes, right? Silly, irresponsible user that I am, I probably should have read those, shouldn't I? Here they are: https://octoprint.org/blog/2020/11/30/new-release-1.5.0/

Well, I did read them. And yes, there's the bold language: "Access Control is now mandatory and no longer can be disabled." and then the link to the guide: "See this guide for a way to have OctoPrint log you in automatically when connecting from an internal IP."

This language initially led me to assume that only private network IP addresses (10.0.0.0/8, 172.16.0.0/12, 192.168.0.0/16, 127.0.0.1) could be configured this way, and so I first dismissed that solution. I was not looking for a solution to allow **some** IP addresses access. I was looking for a solution to  grant **all** IP addresses access, and the language in your documentation (well, your release notes) did not convey this at all, even if the solution was present.

It's true, that guide held the solution. However, given the language used, it was not at all clear that it did so, and this let me on a frustrating wild goose chase.

cp2004, 2020-12-16:
I'll update those docs then, thanks for pointing it out 🙂

cp2004, 2020-12-16:
PR #3891 addresses the documentation errors.

I hope that this is proof that we are trying to be constructive here - humans make mistakes, things get forgotten, hey I can't even remember things I did this morning some days. But lets move on, when the PR is merged then the docs will be correct.

Please, I understand the frustration but keep the conversation constructive. This is an open source project, if at any time you think something needs clarifying then feel free to propose the changes. We are nice people, creating free software for you to use, if something changes then "chances are high there's a valid reason, and it's not pure malice and spite from the OSS maintainer" trying to ruin your day.

muchsimpler3d, 2020-12-16:
Thank you for this pull request... and... 

Actually, let me lead with that, and then move on to the rest: Thank you. Thank you for helping to build a wonderful piece of software such as Ocotoprint. I use it every day, I don't pay for it, and up until recently it worked absolutely beautifully. That took a lot of long hours, thankless work, and you're just giving it away. Believe it or not, I do understand how much effort has gone into OctoPrint, and how much effort it takes to answer people like me on github issues.... a whole, whole lot. So, thank you, I do appreciate it. 

I wouldn't be so completely over-the-top irate, if OctoPrint wasn't such an amazing piece of software, which I relied on so heavily, and worked so well up until now.

When you say:  "chances are high there's a valid reason, and it's not pure malice and spite from the OSS maintainer" I will readily acknowledge that no malice is involved, and that the reason is "valid" using your internal logic. In fact, I am well aware of the internal logic: You don't want users to shoot themselves in the foot. However, I have a very deep and profound belief that those of us users who urgently, fervently wish to remove our feet, who have been previously assured that such a thing is possible, should absolutely be permitted to do so without the rules changing on us. This is a difference of **opinion** not **fact**, and I am fairly certain we will never see eye-to-eye on this. 

No matter how many users systems were seriously completely compromised by unsecured OctoPrint configurations I believe that removing this existing feature is a terrible and fundamentally user-hostile decision. I am, however, willing to acknowledge that it is not **intended** as such.

So, back to the documentation, and the pull request: While updating the documentation to reflect that disabling access control is no longer possible is certainly a good thing, given that it now reflects reality properly. However, it doesn't give any pointers to anyone who does, in fact, wish to disable access control or find work-arounds.

To the best of my knowledge (please, correct me if I am wrong), the best current solution to  approximate fully and completely deactivating access control is to use the autologin feature, specifying the subnets that match all IP addresses. I too, want to be constructive rather than bogged down in vitriol, and the constructive action that I would like to see is for this solution, the next-best thing to actually disabling access control, to be included in the documentation, albeit with a big, red, bold warning "ONLY DO THIS IF YOU ARE ABSOLUTELY SURE YOU WANT THIS" right next to it. That would, in my mind, be the best thing that could happen, and the best constructive action that could be taken on this issue.

I believe that you will never, ever agree to including something like this in the documentation for the exact same reason you removed the access control in the first place. I understand and respect that this is a difference of opinion that is never going to change.

That said, I am going to post everywhere I can, instructions on how to disable access control in OctoPrint 1.5+ because you refuse to include such instructions in the primary documentation. I believe users have a right to be able to do this if they want and I want to help others, like myself, to ensure that they know this is an option. Just as I am willing to acknowledge that you did not make the change out of genuine malice, please understand that I am not doing this out of malice either. I just think that users have a fundamental right to control the software on their own devices as they see fit.

cp2004, 2020-12-16:
> That said, I am going to post everywhere I can, instructions on how to disable access control in OctoPrint 1.5+ because you refuse to include such instructions in the primary documentation.

Hey, again, I have not refused to include auto login local config in *that* section of the docs - it's there under the config.yaml section too, though I appreciate it is not obvious. Generally we try and keep hand holding guides out of the core documentation, but maybe a link to the AAL post (which is under the 'guides' section of the forums)

**No one had even suggested this before** - so thank you for the suggestion. I will take it into consideration.

Instead of posting *everywhere* to navigate this, you could put it in the documentation yourself? Or you could comment your specific suggestions on the open PR (#3891) as to what could be better **for you**.

I was reading through this post and mostly agreed with the points you made until the final paragraph. Not to sound like a broken record, but **be constructive!** If you want something like this documentation to be done differently, then suggest some changes to it rather than go out of your way to undermine the effort we put into OctoPrint. I understand you are frustrated, and that you are not doing this for personal reasons but *come on.*

The way I *now* interpret your comment, having read that final statement is '**thank you for all the great work** that you do with OctoPrint, but now please **understand that I am going to make your life harder** because **I don't want to collaborate** on a good solution and instead will go out of my way to make things worse'.

And as for the whole 'users have rights' to disable things, well I would like to point out that you are completely free to use OctoPrint 1.4.2 for the rest of your life with access control disabled, to fork and maintain your own version of OctoPrint without access control, or not use OctoPrint at all. Having a 'right' to some function of **free open source software** is not a thing, and even if it was, nobody has forced you to use this so they have therefore not infringed upon any rights you may believe you are entitled to whatsoever.

muchsimpler3d, 2020-12-16:
Perhaps I assumed too much, and if so I apologize.

It comes down to this: Are you willing to include anywhere in the documentation (and it can be at the bottom, two levels deep), a description that says that if you are REALLY SURE that you want to disable access control, you can use the 0.0.0.0/0 and ::/0 subnets to effectively disable access control?

That is my specific suggestion, that somewhere lists this solution, with the appropriate warnings as big and scary as you deem necessary, and to preface this information with a detailed description of how to configure the autologin feature to allow only the local network if that's all you need. So long as there, at the end, you finally include how to specify these all-inclusive subnets if that's what a user really, truly, genuinely wants. I would be more than happy (overjoyed, even) to create such a pull request, with this language and these warnings if there is any chance, whatsoever that you would accept it.

The only reason I will post solution elsewhere is that I suspect you will refuse include this useful documentation about your own software in your official documentation. Am I wrong? 

I genuinely, honestly hope that I am wrong, that would be wonderful, and I would have a pull request for you by tomorrow.

cp2004, 2020-12-16:
Since I am not the actual maintainer of this project I can't comment on whether it would be accepted, and beginning yesterday Gina is on a Christmas vacation (well deserved, mind you, this year has been *incredibly* busy for OctoPrint) we might not get that response.

muchsimpler3d, 2020-12-16:
Completely understandable. I will refrain from posting about this information anywhere new until the second week of January 2021 or until I get a response here that suggests the correct solution would be welcome in the official documentation with the appropriate warnings (which, as I've already mentioned, I strongly doubt).

kantlivelong, 2020-12-16:
To put it plainly. We are not going to document how to circumvent a requirement. The ability to do so is present with the options provided.

muchsimpler3d, 2020-12-18:
@kantlivelong That is exactly what I assumed and expected, above. 

However, do you see the kafka-esque absurdity in how you phrased that? "We are not going to document how to circumvent a requirement. The ability to do so is present with the options provided." So, you refuse to fully, accurately document the features you built into your own software, and then get upset when someone else chooses to do so.

I really, genuinely wish I could be constructive here. Constructive implies that there was a way forward, or some form of compromise, but I don't see one. Here is a solution that I absolutely require (Limited subnets were NOT an acceptable substitute in my setup, I NEEDED the 0.0.0.0/0 universal subnet solution). I wish to share that, to help others in my exact situation. I am being told, over and over that you are unwilling to condone specifically what I set out to do. Therefore, I see no way to be constructive or to compromise. If you see a constructive, compromise solution, I would be very open to it.

I am very frustrated here because you seem adamantly opposed to sharing a useful solution to an important problem, with your exceptional software. 

This is what I'm hearing: How dare, how DARE I even contemplate documenting how to accomplish what I want to do with open source software, when no other guide to do so exists? What kind of monster puts in the time and effort to publicly share a solution to the problem that he has to spent significant time solving in the hopes no one else will go through the same? After being rebuffed from contributing to official documentation (which, as outlined plainly above I was/am willing to do) the absolute arrogance of posting such useful, accurate information without the blessing of the original author is just beyond belief, isn't it?

Therefore, I will, on my own terms, without your permission, post my solution in multiple places so that those that want to disable access control can do so. And, having exhausted every method of negotiation here, I won't feel a bit bad about doing so. I feel that this is, ultimately, the right, responsible thing to do. 

I am well aware that you disagree, and why. I simply don't care.

kantlivelong, 2020-12-18:
You're free to post your solution online.

foosel, 2020-12-18:
I cannot go on vacation can I?

@ucirello I'm open to including a mention of the autologin feature on the access control documentation page, and to also point out that it supports arbitrary subnet specifications. Furthermore, to link to the existing guide from there. And I'm even happy to accept a mention of `0.0.0.0/0` and `::/0` there with a ton of warnings. But I reserve *my* right as maintainer of this project to remove that explicit mention again should it prove to lead to yet another case of security incidents that on to of things *I* and my software get blamed for then, like it has happened in the past.

What I have a problem here is *how* you went about suggesting that change. From the very first moment you've antagonized me - the one who actually made that decision and who also has been making the decisions for almost 8y now - here as someone who actively wants to hurt you and people like you personally.

If you had just come in and gone "Hey, the docs are still incomplete here and it would be nice if you also could make it clearer that the network specs can contain arbitrary subnets, including catchall options, not just local subnets", *absolutely no one* would have had a problem here. 

Instead, you came in and immediately started throwing around accusations and demands in an ultra aggressive tone. The only thing missing for the complete "angry customer" experience would have been you screaming to see the manager and to get your money back. 

This is an open source project. The only things you *can* demand here are outlined in the license that comes with it. Everything else is subject to discussion, evaluation and collaboration. Demands and accusations don't play well with this at all, neither do antagonizing the maintainers or an aggressive attitude in general.

I'll revisit this after my vacation. And now everyone take a deep breath and calm down, the year has been bad enough without having to fight about documentation, a bit of compassion and friendliness goes a long way. Happy Holidays.

jneilliii, 2020-12-18:
Yay, we have your official answer @muchsimpler3d, I look forward to your pull request updating the documentation.

ucirello, 2020-12-18:
@foosel @cp2004 @jneilliii - I really apologize that this request of mine became this bitter exchange. Thank you all for being great.

cp2004, 2021-01-06:
Pending acceptance of #3897 that will probably come up in some searches, since it will be on the access control docs (once approved after the holidays).

The other is a blog post, which while it is occasionally updated for new methods, it is not documentation, so I would disagree with putting all the info there.